tcunit/TcUnit #3: Messaging should be done by an interface

sagatowski, 2018-12-07:
The messaging that is done by ADS today (such as in FB_TestSuite.AddTest()) by calling Tc2_System.ADSLOGSTR() should be implemented by an interface, instead of being hard coded to always go to ADS. If we want to be able to log events such as this in a more general way so that we can log it to other sources (say, file-output, database, or any other future means).

Aliazzzz, 2019-05-18:
Will write a FileFactory, this will also help you with "Export results in Xunit XML format #11".
The factory shall be written in vanilla Codesys using the latest compiler (3.5.14.20 or higher)

Aliazzzz, 2019-12-19:
The new XmlWriter uses the interface also

Aliazzzz, 2019-12-31:
You can consider this point as solved

sagatowski, 2019-12-31:
It's not solved. There are still a lot of direct dependencies made to ADSLOGSTR() (by the AdsMessageQueue) in the code.

Aliazzzz, 2019-12-31:
Now I read your anwser, I see that I have misunderstood the original question... My bad!

closed 2020-12-17 by sagatowski