tcunit/TcUnit #4: Decrease the usage of global variables (singletons)

sagatowski, 2018-12-07:
There are several variables defined as in an GVL-list. Global variables are ugly and generally a poor choice from a software design perspective. Try to decrease them (preferably to zero).

maximbendiks, 2018-12-15:
What do you think about moving all globals to the Runner and share a runner with each instance of a test suite(FB_init)? This way you can create multiple runners and therefore different test groups. Each test group can have its own test result.

sagatowski, 2018-12-15:
Thanks for the suggestions/feedback, highly appreciated.

Yes, the runner seems like the natural place for many of these globals variables.

Is there a difference between a test group and a test suite? In googletest (which I'm using C++) the old nomenclature was test group, which I think nowadays it's more common to talk about a test suite (which is the nomenclature used in TcUnit). Their purpose is the same (i.e. group the test cases together for sharing data and subroutines).

Is there any particular reason for when one would want to have more than one runner? I'm trying to see a situation in where you would want to have a separate runner for different test suites. Maybe...one reason could be that you want to run them with different cycletimes? This I'm wondering however if we are not starting to enter the realm of integration (and not unit) testing if we start to be dependent on the cycletime. Or maybe it could be for the reason that we only want to run certain tests? In this case I think it would be better to have some sort of filtering function that I've seen in other testing frameworks (I've personally however never used these filters... but I'm sure there are good use cases for these). Googletest has an excellent description of these here:
https://github.com/google/googletest/blob/master/googletest/docs/advanced.md#running-a-subset-of-the-tests
Let me know what your thoughts are.

I've just written a CONTRIBUTION.MD and commited it. Please read through it and please let me know if you have any questions.

maximbendiks, 2018-12-15:
As you mentioned it is the nomenclature of C/C++ frameworks that I know of. I am fairly new to TDD, but I thought it could be a good idea to refine the rest results and group seemed like a good name for it. Of course you could create only one runner and define the result printer in it. 
 
I will take a look at the runners of googletest and cpputest. Maybe I will get a better idea.

sagatowski, 2020-12-17:
Not actually an issue but rather a discussion. Closing this.

closed 2020-12-17 by sagatowski