tcunit/TcUnit #10: Remove duplicate code

sagatowski, 2018-12-10:
In several of the assert-methods there is quite a lot of code that is duplicated.
Clean that up so it's using common functions/function blocks for these parts.

maximbendiks, 2018-12-11:
Hi Jakob,
for me there are two kinds of duplication:
1. Assert in the method name: assert.Assert... I would rather have something like assert.int_equal. What do you think?
2. Do you want to keep the explicit Equals_TYPE checks? How about only one equal method with ANY inputs? I see only one disadvantage that you can not use constants any more. Otherwise it can be an advantage because you increase readability.

maximbendiks, 2018-12-12:
Alright, for backwards compatibility sakes I get that. But with ReportResult having two ANYs it is going to be harder to fork. I have an idea which would also decrease duplicate code.
How is it best to contribute? By creating a fork or feature branches? Sorry, I never worked with PRs.

sagatowski, 2018-12-13:
There is an excellent guide available at GitHub:
https://help.github.com/articles/creating-a-pull-request-from-a-fork/

I'm writing a contribution-guide for this project, though I'm not done until end of this week, so you can just go ahead! Thanks!

sagatowski, 2018-12-13:
In the latest commit i have wrongfully written that I have solved this issue, which is not true. What I meant is that I solved Issue #12. This issue is still open.

sagatowski, 2020-12-17:
Not really feasible in a sensible way, so closing this now. If re-created, it should be a discussion of how this could be achieved.

closed 2020-12-17 by sagatowski