tcunit/TcUnit #11: Export results in Xunit XML format

fedepell, 2018-12-13:
Connecting also to #7 it would be nice if the framework could output the results in standard Xunit XML format. This way other integration software (ie. Jenkins you mention) can then easily manage and display the results (ie. for Jenkins with the [Junit Jenkins plugin](https://plugins.jenkins.io/junit)).

I believe this pointers can be useful for the format:
https://gist.github.com/erikd/4192748
https://gist.github.com/nil4/7a3cd9c23835ec6b126fe588e836a2e8
https://github.com/windyroad/JUnit-Schema

sagatowski, 2018-12-13:
Excellent suggestion! Will move this to a new milestone.

Aliazzzz, 2019-06-06:
Dear Jakob,

A file factory could be very useful for generating different types of output, like Xunit XML, JSON or whatever.

I did some research on the this subject and the "Abstract Factory" pattern could be very suitable to implement the filefactory.

However, there are very important details to consider when designing the filefactory as an "Abstract Factory". I found some very interesting information on this topic offline (in some .chm files) and googled for it but I had no luck in finding it quickly. It is written for CODESYS point of view, but I bet it is interoperable with TwinCAT.

> In CODESYS  there are two possibilities of dynamically creating an FB instance:
> 
> A.) By using the operators __NEW and __DELETE and a MemoryPool which has to be parameterized for the respecting application via CODESYS. 
> 
> However this method has a significant disadvantage:
>  The MemoryPool is defined once for the application and all the referenced libraries within the application. But it is not possible to determine how much storage space each library is allowed to claim in the MemoryPool. 
> The coincidental utilization of __NEW and __DELETE with different sized FB instances causes a fragmentation of the MemoryPool. Because of that it is not possible to securely install any system for a non-stop operation throughout the entire year. An occasional reboot of the facility would be necessary to re-establish the MemoryPool. 
> **That is why the MemoryPool and the operators __New and __Delete can only be recommended for applications in very special cases. Especially when dealing with libraries this method should not be used!** 
> 
> B.) By using the library FBFactory there is a second and more reliable possibility to meet the demands of dynamically creating an FB instance. 
>  With the help of the FBFactory Library a static pool (which can be extended into a dynamic pool if needed) for each factory is defined in advance. The management of the memory space is organized in way which makes sure that no fragmentation is possible. Therefore there is no obstacle to a non-stop operation of the system.
> 
> FBFactory Library can be used as a base for building factories. In order to get more information about how to use this library for building factories please download the Test Project which you will find in the documentation of the FBFactory Library

So, I'd like to try and implement the filefactory using the 'Abstract Factory' pattern via method B.
If everything works out, it will be usable in TcUnit as well as CfUnit and function as a central filefactory distribution point.

Please let me know what you think of this proposition.

Aliazzz

sagatowski, 2019-06-10:
This was exactly the pattern I was thinking of for this issue. Stefan Henneken has written an excellent blog post regarding this design pattern:
https://stefanhenneken.wordpress.com/2014/11/16/iec-61131-6-abstract-factory-english/

Which is exactly what I planned to use for this issue.

Although considering that this library will be included as a library in other peoples code, and there seem to be an allergy (for good reasons) against dynamic memory allocation, I'm all in for the file factory though with pre-allocated memory.

Aliazzzz, 2019-08-03:
I integrated an XML writer into TcUnit.
I think it should suffice to produce files like this;

```
<?xml version="1.0" encoding="UTF-8" ?> 
   <testsuites id="20140612_170519" name="New_configuration (14/06/12 17:05:19)" tests="225" failures="1262" time="0.001">
      <testsuite id="codereview.cobol.analysisProvider" name="COBOL Code Review" tests="45" failures="17" time="0.001">
         <testcase id="codereview.cobol.rules.ProgramIdRule" name="Use a program name that matches the source file name" time="0.001">
            <failure message="PROGRAM.cbl:2 Use a program name that matches the source file name" type="WARNING">
WARNING: Use a program name that matches the source file name
Category: COBOL Code Review – Naming Conventions
File: /project/PROGRAM.cbl
Line: 2
      </failure>
    </testcase>
  </testsuite>
</testsuites>
```

Aliazzzz, 2019-08-06:
Hi,

Just been searching for the right file markup as I cannot validate it easily;
It seems the following primer is very handy! We just need to verify this.

https://stackoverflow.com/questions/4922867/what-is-the-junit-xml-format-specification-that-hudson-supports

> ```
> <?xml version="1.0" encoding="UTF-8"?>
> <testsuites disabled="" errors="" failures="" name="" tests="" time="">
>     <testsuite disabled="" errors="" failures="" hostname="" id=""
>                name="" package="" skipped="" tests="" time="" timestamp="">
>         <properties>
>             <property name="" value=""/>
>         </properties>
>         <testcase assertions="" classname="" name="" status="" time="">
>             <skipped/>
>             <error message="" type=""/>
>             <failure message="" type=""/>
>             <system-out/>
>             <system-err/>
>         </testcase>
>         <system-out/>
>         <system-err/>
>     </testsuite>
> </testsuites>
> ```
> Some of these items can occur multiple times:
> 
> - There can only be one `testsuites` element, since that’s how XML works, but there can be multiple `testsuite `elements within the `testsuites `element.
> - Each `properties `element can have multiple `property `children.
> - Each `testsuite `element can have multiple `testcase `children.
> - Each `testcase `element can have multiple `error`, `failure`, `system-out`, or `system-err` children.

Aliazzzz, 2019-12-14:
progress report so far
```
(*
	This function block reports the results from the tests into a xUnit compatible XmlFile
*)
FUNCTION_BLOCK FB_XmlFileFormatter IMPLEMENTS I_TestResultFormatter
```

sagatowski, 2019-12-18:
Note to self, I'm using this template: https://wiki.yoctoproject.org/wiki/QA/xUnit_XML_Template that I will be using in the TcUnit-Runner + Jenkins plugin.
I've marked and selected the XML that the first version will support in the attached file.
[image]

Aliazzzz, 2019-12-26:
**Waiting for the update of TcUnit for "centralised" results;**_

Intermediate results;
```
<?xml version=\"1.0\" encoding=\"UTF-8\"?>
<testsuites>
<testsuite name="PrimitiveTypes" errors="22" tests="44"/>
<testcase name="Test_ANY_Differ_DataType"/>
<failure message="Values differ">Assertion failed</failure>
<testcase name="Test_BOOL_Differ"/>
<failure message="Values differ">Assertion failed</failure>
<testcase name="Test_BYTE_Differ"/>
<failure message="Values differ">Assertion failed</failure>
<testcase name="Test_DATE_AND_TIME_Differ"/>
<failure message="Values differ">Assertion failed</failure>
<testcase name="Test_DATE_Differ"/>
<failure message="Values differ">Assertion failed</failure>
<testcase name="Test_DINT_Differ"/>
<failure message="Values differ">Assertion failed</failure>
<testcase name="Test_DWORD_Differ"/>
<failure message="Values differ">Assertion failed</failure>
<testcase name="Test_INT_Differ"/>
<failure message="Values differ">Assertion failed</failure>
<testcase name="Test_LINT_Differ"/>
<failure message="Values differ">Assertion failed</failure>
<testcase name="Test_LREAL_Differ"/>
<failure message="Values differ">Assertion failed</failure>
<testcase name="Test_LTIME_Differ"/>
<failure message="Values differ">Assertion failed</failure>
<testcase name="Test_LWORD_Differ"/>
<failure message="">Assertion failed</failure>
<testcase name="Test_REAL_Differ"/>
<failure message="">Assertion failed</failure>
<testcase name="Test_SINT_Differ"/>
<failure message="">Assertion failed</failure>
<testcase name="Test_STRING_Differ"/>
<failure message="">Assertion failed</failure>
<testcase name="Test_TIME_Differ"/>
<failure message="">Assertion failed</failure>
<testcase name="Test_TIME_OF_DAY_Differ"/>
<failure message="">Assertion failed</failure>
<testcase name="Test_UDINT_Differ"/>
<failure message="">Assertion failed</failure>
<testcase name="Test_UINT_Differ"/>
<failure message="">Assertion failed</failure>
<testcase name="Test_ULINT_Differ"/>
<failure message="">Assertion failed</failure>
<testcase name="Test_USINT_Differ"/>
<failure message="">Assertion failed</failure>
<testcase name="Test_WORD_Differ"/>
<failure message="">Assertion failed</failure>
<testsuite name="AssertTrueFalse" errors="2" tests="4"/>
<testcase name="AssertThatINTsAreEqual"/>
<failure message="INTs are equal">Assertion failed</failure>
<testcase name="AssertThatWORDsAreEqual"/>
<failure message="">Assertion failed</failure>
<testsuite name="AssertEveryFailedTestTwice" errors="1" tests="1"/>
<testcase name="TwiceAssertCall"/>
<failure message="Not equal ANY">Assertion failed</failure>
<testsuite name="CreateFourTestsWithSameName" errors="0" tests="1"/>
<testsuite name="ArrayPrimitiveTypes" errors="38" tests="57"/>
<testcase name="Test_BOOL_Array_DifferInSize"/>
<failure message="">Assertion failed</failure>
<testcase name="Test_BOOL_Array_DifferInContent"/>
<failure message="">Assertion failed</failure>
<testcase name="Test_BYTE_Array_DifferInSize"/>
<failure message="">Assertion failed</failure>
<testcase name="Test_BYTE_Array_DifferInContent"/>
<failure message="">Assertion failed</failure>
<testcase name="Test_DINT_Array_DifferInSize"/>
[... 114 lines cut ...]
<testcase name="TwiceAssertCall_Arrays"/>
<failure message="">Assertion failed</failure>
<testsuite name="AnyToUnionValue" errors="1" tests="24"/>
<testcase name="Test_STRING"/>
<failure message="">Assertion failed</failure>
<testsuite name="MultipleAssertWithSameParametersInSameCycleWithSameTest" errors="1" tests="1"/>
<testcase name="Assert_SeveralTimes"/>
<failure message="Values differ">Assertion failed</failure>
<testsuite name="MultipleAssertWithSameParametersInDifferentCyclesButWithDifferentTests" errors="3" tests="3"/>
<testcase name="Assert_SeveralTimes"/>
<failure message="Values differ">Assertion failed</failure>
<testcase name="Assert_SeveralTimesAgain"/>
<failure message="Values differ">Assertion failed</failure>
<testcase name="Assert_SeveralTimesAgainAgain"/>
<failure message="Values differ">Assertion failed</failure>
<testsuite name="MultipleAssertWithSameParametersInDifferentCyclesAndInSameTest" errors="1" tests="1"/>
<testcase name="Assert_SeveralTimes"/>
<failure message="Values differ">Assertion failed</failure>
<testsuite name="AdjustAssertFailureMessageToMax252CharLengthTest" errors="0" tests="3"/>
<testsuite name="EmptyTestSuite" errors="0" tests="0"/>
<testsuite name="CheckIfSpecificTestIsFinished" errors="0" tests="1"/>
<testsuite name="WriteProtectedFunctions" errors="0" tests="20"/>
<testsuite name="TestFinishedNamed" errors="0" tests="0"/>
</testsuites>
```

fedepell, 2019-12-27:
Just minor comments (comparing with a google test XML generated):

`<?xml version=\"1.0\" encoding=\"UTF-8\"?>`
Are these quotes escapes really needed? (gtest doesn't use them)

`<testsuites>`
Google test also adds a summary, which is then used by Jenkins (or other tools):
`<testsuites tests="6" failures="0" disabled="0" errors="0" timestamp="2019-12-20T11:48:18" time="0" name="AllTests">`
Would that be easy to add? (the counters for sure, maybe not the time, and the name of the test container so if you have many you can divide better when displaying)

Gtest also adds a few more things for each test, not sure how vital but some may be trivial to add:
`<testcase name="send_ESTelemetric" status="run" time="0" classname="ESCommandTest" />`
I suppose the status would be trivial at least.

On the errors:
`<testsuite name="MultipleAssertWithSameParametersInSameCycleWithSameTest" errors="1" tests="1"/>`
Google test seem to use failure here:
`<testsuite name="ESCommandTest" tests="6" failures="0" disabled="0" errors="0" time="0">`
Digging more I found this differentiation [here](https://www.baeldung.com/junit-failure-vs-error):
>  Both failure and error in JUnit tests indicate an undesired situation, but their semantics are different. Failures notify of an invalid test result, errors indicate an unexpected test execution.
So I would say _failure_ should be the right case?

sagatowski, 2019-12-27:
@fedepell Agree on the comments, though I don't understand the purpose of the "Status". I thought the purpose of writing the file/reporting the results was only once the tests were finished running?

sagatowski, 2019-12-27:
@Aliazzzz Could you synchronize so that the output of your writer is equal to the TcUnit-Runner version (according to QA/xUnit XML Template in my previous comment above)?

fedepell, 2019-12-27:
@sagatowski  indeed looking at the schema is not there, maybe just G being overdescriptive :) 
My guts feeling is that you may get for example also aborted or disabled in that line. (and indeed aborted is also one of the values defined [here](https://junit.org/junit5/docs/5.0.1/api/org/junit/platform/engine/TestExecutionResult.Status.html) but disabled is not)

Aliazzzz, 2019-12-27:
Yes, synchronizing the output with the schema has always been the intention. As stated: "Any arbitrary xml tag or parameter can be added as long as we have the accompanying data from the test.

Aliazzzz, 2019-12-31:
Here is a real world output for the Verifier;

<?xml version="1.0" encoding="UTF-8"?><testsuites failures="111" tests="115"><testsuite errors="22" id="0" name="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" tests="44"><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_ANY_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_ANY_Differ_DataType"><failure type="ANY">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_BOOL_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_BOOL_Differ"><failure type="BOOL">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_BYTE_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_BYTE_Differ"><failure type="BYTE">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_DATE_AND_TIME_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_DATE_AND_TIME_Differ"><failure type="DATE_AND_TIME">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_DATE_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_DATE_Differ"><failure type="DATE">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_DINT_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_DINT_Differ"><failure type="Array_DINT">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_DWORD_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_DWORD_Differ"><failure type="DWORD">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_INT_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_INT_Differ"><failure type="INT">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_LINT_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_LINT_Differ"><failure type="LINT">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_LREAL_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_LREAL_Differ"><failure type="LREAL">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_LTIME_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_LTIME_Differ"><failure type="LTIME">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_LWORD_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_LWORD_Differ"><failure type="LWORD">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_REAL_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_REAL_Differ"><failure type="REAL">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_SINT_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_SINT_Differ"><failure type="SINT">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_STRING_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_STRING_Differ"><failure type="STRING">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_TIME_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_TIME_Differ"><failure type="TIME">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_TIME_OF_DAY_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_TIME_OF_DAY_Differ"><failure type="TIME_OF_DAY">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_UDINT_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_UDINT_Differ"><failure type="UDINT">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_UINT_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_UINT_Differ"><failure type="UINT">Values differ</failure></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_ULINT_Equals"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.PrimitiveTypes" name="Test_ULINT_Differ"><failure ty [... cut ...] name="Test_WRITE_PROTECTED_UINT"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.WriteProtectedFunctions" name="Test_WRITE_PROTECTED_USINT"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.WriteProtectedFunctions" name="Test_WRITE_PROTECTED_ULINT"></testcase><testcase classname="CODESYS_Control.Application.PRG_TEST.WriteProtectedFunctions" name="Test_WRITE_PROTECTED_WORD"></testcase></testsuite><testsuite errors="0" id="0" name="" tests="0"/></testsuites>

Aliazzzz, 2019-12-31:
```
<?xml version="1.0" encoding="UTF-8" ?>
<!-- from https://svn.jenkins-ci.org/trunk/hudson/dtkit/dtkit-format/dtkit-junit-model/src/main/resources/com/thalesgroup/dtkit/junit/model/xsd/junit-4.xsd -->
<xs:schema xmlns:xs="http://www.w3.org/2001/XMLSchema">

    <xs:element name="failure">
        <xs:complexType mixed="true">
            <xs:attribute name="type" type="xs:string" use="optional"/>
            <xs:attribute name="message" type="xs:string" use="optional"/>
        </xs:complexType>
    </xs:element>

    <xs:element name="error">
        <xs:complexType mixed="true">
            <xs:attribute name="type" type="xs:string" use="optional"/>
            <xs:attribute name="message" type="xs:string" use="optional"/>
        </xs:complexType>
    </xs:element>

    <xs:element name="properties">
        <xs:complexType>
            <xs:sequence>
                <xs:element ref="property" maxOccurs="unbounded"/>
            </xs:sequence>
        </xs:complexType>
    </xs:element>

    <xs:element name="property">
        <xs:complexType>
            <xs:attribute name="name" type="xs:string" use="required"/>
            <xs:attribute name="value" type="xs:string" use="required"/>
        </xs:complexType>
    </xs:element>

    <xs:element name="skipped" type="xs:string"/>
    <xs:element name="system-err" type="xs:string"/>
    <xs:element name="system-out" type="xs:string"/>

    <xs:element name="testcase">
        <xs:complexType>
            <xs:sequence>
                <xs:element ref="skipped" minOccurs="0" maxOccurs="1"/>
                <xs:element ref="error" minOccurs="0" maxOccurs="unbounded"/>
                <xs:element ref="failure" minOccurs="0" maxOccurs="unbounded"/>
                <xs:element ref="system-out" minOccurs="0" maxOccurs="unbounded"/>
                <xs:element ref="system-err" minOccurs="0" maxOccurs="unbounded"/>
            </xs:sequence>
            <xs:attribute name="name" type="xs:string" use="required"/>
            <xs:attribute name="assertions" type="xs:string" use="optional"/>
            <xs:attribute name="time" type="xs:string" use="optional"/>
            <xs:attribute name="classname" type="xs:string" use="optional"/>
            <xs:attribute name="status" type="xs:string" use="optional"/>
        </xs:complexType>
    </xs:element>

    <xs:element name="testsuite">
        <xs:complexType>
            <xs:sequence>
                <xs:element ref="properties" minOccurs="0" maxOccurs="1"/>
                <xs:element ref="testcase" minOccurs="0" maxOccurs="unbounded"/>
                <xs:element ref="system-out" minOccurs="0" maxOccurs="1"/>
                <xs:element ref="system-err" minOccurs="0" maxOccurs="1"/>
            </xs:sequence>
            <xs:attribute name="name" type="xs:string" use="required"/>
            <xs:attribute name="tests" type="xs:string" use="required"/>
            <xs:attribute name="failures" type="xs:string" use="optional"/>
            <xs:attribute name="errors" type="xs:string" use="optional"/>
            <xs:attribute name="time" type="xs:string" use="optional"/>
            <xs:attribute name="disabled" type="xs:string" use="optional"/>
            <xs:attribute name="skipped" type="xs:string" use="optional"/>
            <xs:attribute name="timestamp" type="xs:string" use="optional"/>
            <xs:attribute name="hostname" type="xs:string" use="optional"/>
            <xs:attribute name="id" type="xs:string" use="optional"/>
            <xs:attribute name="package" type="xs:string" use="optional"/>
        </xs:complexType>
    </xs:element>

    <xs:element name="testsuites">
        <xs:complexType>
            <xs:sequence>
                <xs:element ref="testsuite" minOccurs="0" maxOccurs="unbounded"/>
            </xs:sequence>
            <xs:attribute name="name" type="xs:string" use="optional"/>
            <xs:attribute name="time" type="xs:string" use="optional"/>
            <xs:attribute name="tests" type="xs:string" use="optional"/>
            <xs:attribute name="failures" type="xs:string" use="optional"/>
            <xs:attribute name="disabled" type="xs:string" use="optional"/>
            <xs:attribute name="errors" type="xs:string" use="optional"/>
        </xs:complexType>
    </xs:element>

</xs:schema>
```

sagatowski, 2019-12-31:
Guess the source of the last comment is this XSD:
https://github.com/junit-team/junit5/blob/master/platform-tests/src/test/resources/jenkins-junit.xsd

(putting it there as a reminder so I can use it myself when verifying the functionality of TcUnit-Runner)

Aliazzzz, 2019-12-31:
Excellent, we should adhere to the v5 xsd, as mine is v4..
I expect it to be backward-compatible, but I'll test it asap anyway.

To be continued

sagatowski, 2020-01-03:
The XML-creator part of TcUnit-Runner is finished and working. It produced the following artifact:

[testresult_tcunitrunner.xml.txt](https://github.com/tcunit/TcUnit/files/4019461/testresult_tcunitrunner.xml.txt)

I verified it against the following XSD:
https://github.com/junit-team/junit5/blob/master/platform-tests/src/test/resources/jenkins-junit.xsd

And it's all fine

SoftwareQualityLabPirklbauer, 2020-01-13:
Hi
Since I am really interested in this feature, I have a question to you.
I have seen, that this feature will be implemented in the next release 1.1.0.0.

Is there a guide how the XML-creator will be used in TwinCat (FB, Function, ...)?

Best regards

sagatowski, 2020-01-13:
Hi!

The xUnit XML-file creation is done by a program (still under writing/documentation, will be released as open source as well) that does the integration of TwinCAT/TcUnit into a CI/CD tool (Jenkins). This program is called "TcUnit-Runner", and is executed by Jenkins and does:
-	Load the TwinCAT project
-	Load the correct version of TwinCAT, based on what the project was made in
-	Look for/Find the TcUnit task
-	Load only that task (not the other tasks, if there are any)
-	Compile that specific task/program in the project
-	Activate the configuration of that build on a target
-	Run all the tests
-	Report test results to xUnit XML file

In a future release (after 1.1) of TcUnit it will also be possible to create the XML-file directly in TcUnit, though the supported way of doing this in 1.1 will be through the TcUnit-Runner

SoftwareQualityLabPirklbauer, 2020-01-13:
Thanks for your quick response

Sounds great!
So, I am really looking forward to 1.1

sagatowski, 2020-01-13:
@SoftwareQualityLabPirklbauer 
N.P! Other than the software there will be plenty of documentation released together with 1.1, and unfortunately this is what's taking most of the time now :-)

Aliazzzz, 2020-01-13:
I made a PR for the new JUnitXmlPublisher which is embedded within TcUnit. The code is written in such a way that it is easily extensible if needed.

DominicOram, 2020-01-15:
@sagatowski This is great! your description of `TcUnit-Runner` actually matches something we've been working on internally for testing. I have a couple of questions/suggestions about it:
1. I assume it's based on the `Automation Interface`? In which case it's C# based?
2. `Compile that specific task/program in the project` I was actually compiling all code as if the project fails to compile that's an error we should let Jenkins know about
3. We have additional tests we perform where we run up a task and confirm that we can communicate with the PLC using `ADS` from a different piece of software. Essentially the job of the test runner here is to start a task and then stop it when I say I'm happy.

Are use cases 2 and 3 something you would be happy being in `TcUnit-Runner` as options? If so we'd be happy to do the implementation ourselves.

sagatowski, 2020-01-15:
> @sagatowski This is great! your description of `TcUnit-Runner` actually matches something we've been working on internally for testing. I have a couple of questions/suggestions about it:
>
> 1. I [...]

@DominicOram That's great! Does it mean you've basically already done a "TcUnit-Runner" for TcUnit?

To answer your questions:

1. Correct/correct
2. Thinking about it, it makes more sense to compile the whole project to see it compiles, and only run the test if it does. Regarding the running it's necessary to run only the TcUnit-task as some users might have test cases in full-blown programs (not libraries) where there are tasks linked to I/Os etc (which will not be able to execute then)
3. Can you elaborate this further? 

You're more than welcome to contribute to the TcUnit-Runner. It's a private repo I've been experimenting with but I'll make sure to make it public on the TcUnit-site quite soon.

DominicOram, 2020-01-15:
> Does it mean you've basically already done a "TcUnit-Runner" for TcUnit?

Not really, we've only just started looking into tcUnit. Currently my test runner just confirms that everything compiles and then performs the testing described in 3). Adding tcUnit in was the next thing I was going to look at doing. The code for this is in https://github.com/ISISComputingGroup/BeckhoffTestRunner if you're interested.

> Regarding the running it's necessary to run only the TcUnit-task as some users might have test cases in full-blown programs (not libraries) where there are tasks linked to I/Os etc (which will not be able to execute then)

Agreed.

> Can you elaborate this further?

Happy to elaborate further but this feels like getting well off topic for this issue. Maybe I could add an issue to the `TcUnit-Runner` repo when it's public to describe the situation in depth? In the mean time there is some documentation [here](https://github.com/ISISComputingGroup/ibex_developers_manual/wiki/Beckhoff#continuous-integration) that describes the system. The program described as `Automation Tools` in that documentation is the one that sounds similar to `TcUnit-Runner`. The documentation is aimed at our internal developers though so may raise more questions than answers...

sagatowski, 2020-01-15:
> Not really, we've only just started looking into tcUnit. Currently my test runner just confirms that everything compiles and then performs the testing described in 3). Adding tcUnit in was the next [...]

@DominicOram 
The BeckhoffTestRunner seems like a very interesting project. I'll look deeper into it! Just by looking at the "continous integration" picture of the documentation we seem to struggle with the same problems.

I'll make sure to make the TcUnit-Runner repo public as soon as possible. I would highly appreciate feedback/issues for the TcUnit-Runner. The code is "experimental" to say at least (limited error handling, not all VS/TwinCAT versions are handled correctly, etc), so contributions/ideas are more than welcome. I'm happy that you seem to work with the same challenges I've been struggling with, so I'll make sure to ping you here and write the link to the TcUnit-Runner once it's public.

Aliazzzz, 2020-01-18:
Update: added a quickfix to enable the JUnitXmlPublisher.
I tested it under TwinCAT v4022.30 and now **works**.

Output path defaults to : "C:\testresults.xml' and can be changed via a parameter.

sagatowski, 2020-04-06:
Parts of this functionality is now merged into the main by the PR #91.
https://github.com/tcunit/TcUnit/pull/91

sagatowski, 2020-09-05:
Finally implemented. To state this was a major task would be an understatement.
See:
- https://github.com/tcunit/TcUnit/releases/tag/1.1.0.0
- https://github.com/tcunit/TcUnit-Runner

closed 2020-09-05 by sagatowski