tcunit/TcUnit #20: Tests with timers inside the FB

Seversonic, 2019-02-13:
Currently tests only run for one scan. 

This makes running tests on FBs with internal timers difficult. There is a workaround, but it would be great to have tests mean to run for a minimum time.

Robinsir, 2019-03-11:
I have the same problem.

sagatowski, 2019-03-11:
I'm working on a variant of TEST() where you can define a time that the test shall run. Until that is finished, and equally a good solution, I've created an example of how this can be accomplished with the current version of TcUnit.

I've created an example TwinCAT-project for you that demonstrates how this can be done.
https://tcunit.org/temp/TimedTest.zip

I've created a simple function block that has a timer built-in (FB_ToBeTested), which simulates the problem. The trick here is that the test-suite is not allowed to return all tests finished until the function block has had enough time to run. I've basically just created a timer in the test-suite that doesn't elapse until the timer in the main-function block has elapsed. The only problem I see with this is that the test-suite needs to know how long the FB-under-test has to run before finishing (which it most likely needs to know anyway). Another alternative is to only consider the test finished if the function-block under test has returned some sort of returnvalue indicating that it is finished.

Robinsir, 2019-03-12:
> I'm working on a variant of TEST() where you can define a time that the test shall run. Until that is finished, and equally a good solution, I've created an example of how this can be accomplished [...]

That's great. Thank you！

sagatowski, 2019-08-05:
This functionality is redundant now that the new API for the framework requires to do TEST_FINISHED() when a test is finished, which means the user decides himself when each individual test is finished.

closed 2019-08-05 by sagatowski

YokMenns, 2020-01-08:
Hi sagatowski,
this is great! It looks like the example mentioned here with timer isn't compatible with the lastest new TcUnit (1.0.0.0). Is that true? or maybe i make somtehing false? Thank you!

sagatowski, 2020-01-08:
> Hi sagatowski,
> this is great! It looks like the example mentioned here with timer isn't compatible with the lastest new TcUnit (1.0.0.0). Is that true? or maybe i make somtehing false? Thank you!

Hi YokMenns!
That's correct. That example was made using the beta-API (pre 1.0). With the release of version 1.0 (and later) the API was frozen. I've created a new example for you:
https://tcunit.org/temp/TimedTest_1x.zip

YokMenns, 2020-01-09:
Hi Sagatowski,
thank you for your quick answer. Yesterday, i used the TcUnit 1.0.0.0 and Timer. because of your old example it was very easy to modify it. But thank you for you new example, i will check your code. It is always interesting to see a different mind. Thank you!

YokMenns, 2020-01-16:
Hi sagatowski ,
i convert your actual tcUnit 1.1.0.0 in libray and i try to write my tests with it.
It looks like if i have a counter in the FB and if i define a counter in the FB_Test ( refere to your example with the timer), the system don't show me any testresult more. I mean in the lg i see the failed tests but in errors and the other tests as messages but the test about checking the counter still missed on the log and also the rsult statistic still missed. i know the tcUnit isn't release, but maybe it is better to know it now. Or maybe i make something false. Please can you check it. Thank you

sagatowski, 2020-01-16:
Hi @YokMenns 
I'm not sure I follow what you mean.
1. Do you mean running the TimedTest_1x.zip with the current 1.1. in the master branch?
2. I tested it and it works fine. Did you wait the 10s for the results to display?

/ J

YokMenns, 2020-01-16:
Hi Jacob,
yes i mean the TimeTest_1x.zip with the current 1.1. in the master branch, but replace the Timer by a counter like below as example.

FUNCTION_BLOCK FB_ToBeTested
VAR_INPUT
    x : REAL;
    y : REAL;
END_VAR
VAR_OUTPUT
    sum : REAL;
END_VAR
VAR
    //ThisIsATimer : TON := (PT := T#10S, IN := TRUE);
iCounter : INT := 100;
END_VAR
#############################
iCounter := iCounter + 1;
IF iCounter = 100 THEN
    sum := x + y;
ELSE
    sum := 0;
END_IF
##################################
and for FB_ToBeTested_Test i implement another counter for example iCounter_Test.
I Tested it with the 1.0.0.0. But now with the current 1.1. in the master branch, it doesn't work.
YokMenns

sagatowski, 2020-01-16:
Hi @YokMenns 
I tested the example you provided with both TcUnit 1.0 (release) and 1.1 (master branch) and in both cases it worked (although the example you provided FAILED, obviously because it only returns the correct value whtn icounter = 100, which is not the case when the timer times out).

YokMenns, 2020-01-16:
Hi sagatowski,
thank you for your investigation. I am trying to check it again now. But it looks like my TwinCAT doesn't work. I check it again and i let you know. Thank you

YokMenns, 2020-01-17:
Hi Hi sagatowski,
it works fine with the 1.1 (master branch). Maybe i had a mistake on my code.
I cehck both oh them and both of the tcUnit work. Thnak you.
Tests finisched result is fine now. only failed tests and statistic occuered as error. well done. great job.