tcunit/TcUnit #27: Multiple assert instances in one test suite causes issues

claytonketner, 2019-06-06:
Instantiating >1 instances of `FB_Assert` in your test suite causes incorrect test pass/fail reporting, depending on the order they are instantiated. Any instances after the first will report incorrectly, meaning, if the assert fails:

- The test will still show in the final test summary as passing
- The instance path in the log output is empty: `FAILED TEST '@testName', ...`

claytonketner, 2019-06-06:
Furthermore, if you instantiate `FB_Assert` in the `VAR` of your test method, it causes a stack overflow.

sagatowski, 2019-06-07:
Good catch. As you say, normally there is no reason for an user to create more than one instance of an assert-FB. However, doing an instantiation of fb-assert of a test-method is quite common in how you do it in other (non-plc) unit-testing frameworks. Although I'm not entirely sure this could (or should) be possible to do in TcUnit, it should at least not crash and the user should get a notification of that the instantiation of the assert-fb should be done in the header of the testsuite. Any other suggestion is welcome!

sagatowski, 2019-06-15:
As we've abandoned the FB_Assert function block altogether (included in the FB_TestSuite), it's no longer possible to create multiple instances of the FB_Assert FB. In other words, this is solved in the commits 1ca08abb12dc0bdadb7161ef079616b0b596ce3d, 48b2a750e66a13b78b6bc0425b6a4e903c833ab6, 826045f00371183993cf18e92a142138dc3d6820.

closed 2019-06-15 by sagatowski