tcunit/TcUnit #30: An empty test suite causes the test report to never show

claytonketner, 2019-06-18:
This is likely due to the runner thinking it is still running tests.

sagatowski, 2019-06-18:
Sorry I'm not 100% sure I fully understand. You mean empty test-suite in that it has no test cases and no asserts? If so, what would be the best suggestion? Maybe checking that if we execute a test-suite, and we detect a new test-suite without running any tests, we should output an error message? Thanks for this find!

Aliazzzz, 2019-06-18:
Hi,

I also do not follow this.

Maybe it would be a good idea to upload a small code example which yields this bug.
Otherwise we keep guessing.

From there we can assess whether this issue is a bug or a user problem or not.

Aliazzz

claytonketner, 2019-06-18:
Yeah apologies for the lack of writeup. I updated the OP with more detail.

sagatowski, 2019-06-19:
Thanks for the clarification. I'll look at the issue.

slacAWallace, 2019-07-26:
Experiencing something similar. Bumped to 0.9, refactored and the test summary stopped displaying. I have two empty test suites.

sagatowski, 2019-07-26:
> Experiencing something similar. Bumped to 0.9, refactored and the test summary stopped displaying. I have two empty test suites.
> 
> edit
> Yeap, definitely the same thing. I made sure all tests were FINISHED and the summary reappeared.

Thanks Alex. I'll try to look into this as soon as possible (and for sure before the 1.0 release).

sagatowski, 2019-07-26:
Solved in commit 094faf7f. If no test-cases are declared in a test-suite, this test-suite is simply ignored. The way this is detected is that if we are on the second cycle (or later) and no tests have been run for that test-suite in the first cycle, we consider that test-suite as incomplete and simply ignore it (AllTestsFinished = True).

If the user has tests declared (like TEST('ThisIsATest')) but forgets to add TEST_FINISHED() there is nothing the framework can do, because we simply don't know for how long a user wants to run a certain tests before considering it TEST_FINISHED().

sagatowski, 2019-07-26:
> Experiencing something similar. Bumped to 0.9, refactored and the test summary stopped displaying. I have two empty test suites.
> 
> edit
> Yeap, definitely the same thing. I made sure all tests were FINISHED and the summary reappeared.

Alex, you're welcome to clone the latest version (only source, no release yet) and see if it works out for you.

closed 2019-07-26 by sagatowski

slacAWallace, 2019-07-26:
OK I will try it. BTW, 0.9.0 is awesome! Multi-cycle FB unit testing is really helping me out!

Aliazzzz, 2019-07-26:
Hi,

I have also made a very handy multi cycle testframework in CODESYS for CfUnit, should I redo it for TwinCAT? May it helps you even better? Let me know.