tcunit/TcUnit #32: restarting tests after completion

Aliazzzz, 2019-06-26:
Hi,

I have the following scenario;
While writing new testcases I frequently want to verify the new tests.
I understand that the framework is built to just "run once", but this is kind of a dilemma when developing new tests. Is there an easy way to rerun the tests without (cold)resetting the controller? I would like to hear your thoughts on this.

With kind regards,

Aliazzz

sagatowski, 2019-06-26:
You can ignore a test by simply starting the testname with disabled_. Then you don't need to run the tests that are taking a long time. If this was the problem?

Aliazzzz, 2019-06-26:
Nope... It is not the problem.
The 'problem' is when I write new tests, I want to run test-drive these tests usually multiple times...

sagatowski, 2019-06-27:
I'm not sure I follow then. You only want to re-run the tests that are new, and also without doing a restart of the runtime?
Or is it just that you don't want to restart the runtime, though it's fine that all tests are re-run?

Aliazzzz, 2019-06-28:
Bingo! 
Re-run the tests that are new, without doing a restart of the runtime. That is what I ment.

The writing and the tuning of a new test, running the test with cold starting the PLC each time gets "tedious" after a full day... I think I inherited a healthy dose of lazyness.

I have a suggestion. Maybe we can think of a feature which alleviates the dev-cycle. 
It will save some time and effort as the RUN() command simply runs all testssuites once and only once.
This offcourse is good and should not be altered. However, maybe we can add a new function which can start and restart (run/rerun) a certain testsuite (or predefined testsuites) at will. 
RERUN( ptTestSuiteWhichWeWantToRUN, xStart ); 

The function is offcourse only neccesary in debug cycles.
When writing and testing of tests is done, this code should be replaced with RUN();

I'd love to hear your opinion on this

sagatowski, 2019-07-07:
In any case I guess it would be necessary to do a restart of the runtime, if the code is to be altered, no?

What I usually do when I need to re-run test is to simply comment all the test-suites that I know are passing, test the new test-suites, and then uncomment all. Isn't this (in most cases) just as fast as have to create some sort of array with pointers to a test-suite, and make a separate call to them (while re-starting the runtime as well)?

Aliazzzz, 2019-07-07:
The usecase requires no coding, just patience ;-)

Closed

closed 2019-07-07 by Aliazzzz