tcunit/TcUnit #33: Replacing GVL_Constants_TcUnit for a ParameterList

Aliazzzz, 2019-06-28:
Hi there

I'd like to suggest a small 'upgrade' to replace the GVL_Constants_TcUnit with a parameter list.
The advantage of a parameter list is that the parameterlist is still part of the Libray while parameter list values can still be changed if the user wishes (!)

Examples of "parameter list" usage;
[https://stefanhenneken.wordpress.com/2017/08/08/iec-61131-3-parameterbergabe-per-parameterliste/](https://forge.codesys.com/lib/debuglogger/home/Home/) 
And in my own work;
[https://forge.codesys.com/lib/debuglogger/home/Home/](https://forge.codesys.com/lib/debuglogger/home/Home/)
(see subject Change the default Ringbuffer size)

sagatowski, 2019-07-07:
Sounds like a good idea.
By the way, the link to the codesys site requires you to have a codesys account.

Aliazzzz, 2019-07-07:
I have made this change in a fork. 
I will also add some other small updates before I will create the PR.

sagatowski, 2019-07-13:
Solved in commit b035414.

closed 2019-07-13 by sagatowski