tcunit/TcUnit #38: Test context gets lost

slacAWallace, 2019-07-30:
Relating to: #37 

I was trying to use tcunit to test a statechart, using a case statement. I acknowledge this is somewhat outside the normal realm of unit testing. Regardless, using a case statement to change which test is running for a given cycle works, with some caveats, and this leads to a "bug" with tcunit. tl;dr TcUnit does not currently support nested tests, but it can!

Tests apparently register themselves with the test engine, and this registration includes a boolean to indicate the test is complete. If all test suites and tests indicate they are done, when tcunit runs, then tcunit processes the results and sends them for logging ( I think).

If your test suite is using a case statement to remain focused on a single test at a time, your first test might not get registered before tcunit finishes. So to counter this, you can either make sure the first case has a "wrapper" test, or your first test starts in the first case. Then as you move from case to case you indicate each test in that case is finished. Tcunit apparently figures out the test context by the last TEST('') it sees (I think...).

If you use a wrapper-test, tcunit does not have a way of knowing which test context it's in after TEST_FINISHED is called. ie. Tcunit will not know it's back in the wrapper test context.

[Here's the test suite example](https://github.com/slacAWallace/pmps-dev/blob/ecat-comm/pmps-plc/PMPS_Prototype_Arbiter/POUs/TestComponents/FB_BPTM_Test.TcPOU)

sagatowski, 2019-07-31:
Alex, I can't compile the function block (missing DUTs). Do you happen to have a full-working solution with this FB?

slacAWallace, 2019-07-31:
OK, I fixed the solution so it includes the enum that was missing. That link is to the specific fb, but the repo contains the full solution.

Roald87, 2019-08-05:
I wanted to make a new Issue for an enhancement idea I had, but I think you describe it here. Although the title and explanation were initially not very clear to me: I had to read it multiple times, but I think I now know what you mean.

To put it my own words: You want a method to check if a specific test in the current suite has finished or not?

I think this is a very useful option since we also often need tests to run sequentially, i.e. on after the other.

Maybe it is an idea to update the title and your explanation a bit? For example: “Add a way to check if a specific test from the suite has finished” or “Make option to selectively run individual tests”?

sagatowski, 2019-08-05:
Just wanted to let you know that I've started to look into the example code to understand the case

sagatowski, 2019-08-05:
Hi Alex! I've looked into the code but two questions:

1. it seems I'm not looking into the right example I think... because I can't find any TEST_FINISHED() (either named or non-named) in the code. I'm looking in [this example](https://github.com/slacAWallace/pmps-dev/tree/ecat-comm). Where shold I look instead?

2. Could you create some tests for [TcUnit-Verifier](https://github.com/tcunit/TcUnit-Verifier) so that its clear what the intent of the new functionality is and that it is also clear that that functionality is correctly implemented? Once this is done, create a separate PR for TcUnit-Verifier.

slacAWallace, 2019-08-05:
You have the correct repo and branch. `TEST_FINISHED('')` is the API I am using. I could understand why we might not want to change the API at this point, so `TEST_FINISHED_NAMED()` is an alternative.

I will work on the verifier PR.

slacAWallace, 2019-08-05:
You're right, I should include a warning message in either API case.

sagatowski, 2019-08-06:
I noticed I was on the wrong branch of your example code. Bummer.

I have just merged another PR into TcUnit and I'm about to merge this one as well. I will cherry-pick the commit where you've added TEST_FINISHED_NAMED().

Once this is done, could you do a pull of the latest commits, and add the warning on top of that as a separate commit? So we won't have any merge issues.

sagatowski, 2019-08-06:
Solved in commit 4424dad.

I've merged your change. Please do a pull and add the warning.
I've created a separate issue #42 for the warning.

Also make sure to add new tests that verify this functionality in TcUnit-Verifier. I've created an issue for that at:
https://github.com/tcunit/TcUnit-Verifier/issues/5

Make sure all new tests are working according to:
https://github.com/tcunit/TcUnit-Verifier/blob/master/README.md

Make sure all formatting for the new code is according to:
https://github.com/tcunit/TcUnit/blob/master/CONTRIBUTING.md

Thank you for your contribution!

closed 2019-08-06 by sagatowski