tcunit/TcUnit #42: TEST_FINISHED_NAMED should warn if name doesn't exist

sagatowski, 2019-08-06:
Related to issue #38.
If TEST_FINISHED_NAMED is called with a name for a test that doesn't exist, the user should be warned (by standard ADSLOGSTR).

slacAWallace, 2019-08-08:
Started on this tonight. Will test it tomorrow and make pr.

sagatowski, 2019-08-08:
Great Alex, looking forward to your PR!

slacAWallace, 2019-08-08:
@sagatowski What do you think about keeping the bleeding edge of master set to 0.0.0.0 and unreleased and only setting version numbers on tags? I find I keep changing the project version to avoid interfering with my real install of TcUnit. Also with 4024, the released checkbox prevents changes to the code.

Aliazzzz, 2019-08-08:
Hi,

Though the idea sounds good, I’d consider not to do that.
Especially in a big project it is an unwise decision to set the version to a lower or even unused versionnumber then the currently released version as you easily lose track of the version and changes.
We should keep the unreleased version always higher than the released version.

I have got the following empirical library development guidelines on versioning of CODESYS and TwinCAT libraries for you to get an idea on how to use it;
https://forge.codesys.com/forge/community/Developer%20Community%20Documentation/#versioning-of-own-items

With kind regards,

Aliazzz

sagatowski, 2019-08-08:
I'd have to agree on the middle-road between these two. I'll keep the version number to a later one than the latest released/tag, but only set the "released"-flag once it's time for a tag/formal release.

slacAWallace, 2019-08-09:
I wish the git describe output could be used in lieu of a proper tc version, until the released button is checked. :(

sagatowski, 2019-08-19:
Alex, my plan is to release TcUnit version 1.0 this weekend. Do you think you'll have time to finish this + add tests to TcUnit-Verifier prior to that?

Cheers

closed 2019-08-26 by sagatowski