tcunit/TcUnit #44: Unifying visual studio versions across projects

Roald87, 2019-08-12:
In the [`CONTRIBUTING.md`](https://github.com/tcunit/TcUnit/blob/master/CONTRIBUTING.md) you mention that we should use the same Visual Studio version number as the original solution file. I found that there are three different visual studio versions used:

- TcUnit (Visual Studio 2013)
- TcUnit-Verifier_TwinCAT (Visual Studio 14 = Visual Studio 2015)
- TcUnit-Verifier (Visual Studio 15 = Visual Studio 2017)

I happen to have them all installed. Nevertheless it would be a good idea to use one version for all projects.

sagatowski, 2019-08-13:
Good point. Do you know how to downgrade both verifiers to Visual Studio 2013?

Roald87, 2019-08-13:
Not sure, but why would you want to use 2013 and not Visual Studio 2017? I think 2017 also supports all TwinCAT functionality. VS2019 is not officially supported yet I think. 

Can't you open the files in the visual studio version you want to have them and then upload that sln file? Not too much should change I think when you open it in another visual studio version.

sagatowski, 2019-08-13:
For 3.1.4022 VS2017 is not yet fully supported (only partially).

Did you try if that works?

Roald87, 2019-08-13:
Since a few months we made the switch to VS2017 and use 3.1.4022. So far we haven't noticed any lacking features. Beckhoff indeed mentioned that VS2017 supports ~95% of the functionality, whereas VS2015 is fully supported. They didn't share which features were not supported.

I haven't tried to open the projects with 2013 yet.

sagatowski, 2019-08-13:
We tried VS2017 some ~6 months ago, but had several issues where we decided to go for the version that was default included with TwinCAT as it made unifying easier (VS2013, 2015, 21017 and all variants VS community, full, professional etc)

sagatowski, 2019-08-13:
I'm working on the conversion now. Was (of course) much easier in theory than in reality. I think the easiest way is to just re-create both projects in VS2013.

sagatowski, 2019-08-13:
Solved.
All projects are using VS2013 now. Issue was solved by re-creating both projects and painstakingly copying one file at a time and making sure all project properties are correct.
TcUnit-Verifier (TwinCAT and .NET) are tested working on a machine with VS2013.

Also updated [CONTRIBUTING.md](https://github.com/tcunit/TcUnit/blob/master/CONTRIBUTING.md).

closed 2019-08-13 by sagatowski

Aliazzzz, 2019-08-13:
Hi all!

I am glad to hear  that the IDE will stay vanilla installed Beckhoff VS2013. Whenever Beckhoff migrates we should migrate with them seamlessly. For now they provide VS2013 per default.

Closed #44<https://github.com/tcunit/TcUnit/issues/44>.