tcunit/TcUnit #55: TEST_FINISHED() seems to not pass through the complete TcUnit stack

arpitpatel901, 2019-10-09:
In the case when a TEST needs to run multiple scans before it is finished, the TEST_FINISHED() api does not seem to completely exit out of the TcUnit stack frame(has no output in the Output/Error List) window.

[Test code link](https://drive.google.com/file/d/1RTmpwNBIGJwQVclgkq45mXMOf0wmjIul/view?usp=sharing)

sagatowski, 2019-10-10:
Thanks for creating this issue. I will investigate this.

sagatowski, 2019-10-20:
I've looked at the code, and for some reason you are calling TEST_FINISHED twice for each test. For each TEST(''), you are only supposed to call TEST_FINISHED() once.
I removed the two TEST_FINISHED() in the body of FB_AirCylinder_Test, and then the test suite passed through (although with one failed and one successful test).

sagatowski, 2019-11-04:
@arpitpatel901 , could you confirm this indeed solved your issue?

sagatowski, 2019-11-17:
@arpitpatel901, Closing this issue. If you can re-create the issue, please open it again.

closed 2019-11-17 by sagatowski