tcunit/TcUnit #56: Testing Switch case and non-Static data

kumaraswamygaviyappa, 2019-10-16:
Need help in unit testing the Switch case.

In the Below example:
1. Always case:0 is executed as BA default value is 0 and for the next test BA=1 will not be effective .
2. when BA=1 is made explicitly(which is what i wanted), only IF part is executed not ELSE part as OUT_Sollwert default value is 0 and the latest data will not be stored.

I would like to know:
1. Is there any way, i could test this.
2.Could you please share the unit testing example of switch case

`VAR_OUTPUT
	OUT_Sollwert: REAL;
END_VAR

VAR
       BA: INT;
END_VAR

CASE BA OF
	0:
                OUT_Sollwert:= data;
		BA:= 1;
	1:	
		IF(OUT_Sollwert < (Velocity*time)THEN
			OUT_Sollwert:= OUT_Sollwert + (Velocity*time);
		ELSE
			OUT_Sollwert:= Endlage_oben;
                        BA:= 2;
               END_IF
        2:
`

maximbendiks, 2019-10-16:
I guess this is a method, then BA and OUT_Sollwert are both temporary variables. You either have to declare them in your FB or use VAR_INST instead of VAR in the declartion section. For OUT_Sollwert you would need a buffer variable in your FB or VAR_INST and then simply assign its value to your output at the end of your method.

kumaraswamygaviyappa, 2019-10-16:
@maximbendiks Thank you very much for the advice.

This is a FB and as a whit-box tester, i don't have permissions to change the code.
Using 3 input variables, I was generating test cases to cover all **CASE** , **IF** and **ELSE** parts. 
How could i test the **CASE** as **BA** and **OUT_Sollwert** will have new values, and used to decide next **CASE** and to calculate new position respectively.
In **TcUnit Framework**, these are assigned to Default values which is 0 in this case for every function call.

```
FUNCTION_BLOCK FB_DreieckTest

VAR_INPUT
	IN_Start: BOOL;			
	IN_ZykZeit: REAL;			
	IN_v: REAL;				
END_VAR

VAR_OUTPUT
	OUT_Sollwert: REAL;		
	OUT_Endlage: BOOL;		
END_VAR

VAR
	BA: INT;
	Endlage_oben: REAL:= 150.0;		
	Endlage_unten: REAL:= 5.0;		
	a_brems: REAL:= 1000.0;		 
	v_aktuell: REAL;			
END_VAR
___________________________________________________________________________________________________________
CASE BA OF
	0:	
		IF(IN_Start = TRUE)THEN
			BA:= 10;
		END_IF
		OUT_Sollwert:= Endlage_unten;
		OUT_Endlage:= TRUE;
		v_aktuell:= IN_v;
	10:	
		IF(OUT_Sollwert < (Endlage_oben - (IN_v * IN_ZykZeit)))THEN
			OUT_Sollwert:= OUT_Sollwert + (IN_v * IN_ZykZeit);
		ELSE
			OUT_Sollwert:= Endlage_oben;
			a_brems:= 1000;
			BA:= 20;
		END_IF
		OUT_Endlage:= FALSE;
	20:	
		IF(a_brems > 0 )THEN	
			IF(OUT_Sollwert < (Endlage_unten + (v_aktuell + IN_v / (2*a_brems))))THEN		
				v_aktuell:= MAX(v_aktuell - (a_brems * IN_ZykZeit),0);						
			END_IF
		END_IF
		IF(OUT_Sollwert > Endlage_unten)THEN
			OUT_Sollwert:= MAX(Endlage_unten, OUT_Sollwert - (v_aktuell * IN_ZykZeit));
		ELSE
			OUT_Sollwert:= Endlage_unten;
			BA:= 0;
		END_IF
END_CASE
```
**Unit Testing:**
```
TEST('FahrenAufObereEndlage');

fb_DreieckTest(IN_Start:=TRUE,
	IN_ZykZeit:= Input_ZykZeit,
	IN_v:= Input_v,
	OUT_Sollwert=>OUT_Sollwert_Result,
	OUT_Endlage=>OUT_Endlage_Result);
	
AssertEquals_REAL(Expected := OUT_Sollwert_Expected,
                              Actual := OUT_Sollwert_Result,
			      Delta := Delta_Value,
                              Message := 'Test $'FahrenAufObereEndlage$'failed at $'OUT_Sollwert$'');
			 
AssertEquals_BOOL(Expected := OUT_Endlage_Expected,
                                Actual := OUT_Endlage_Result,
                                Message := 'Test $'FahrenAufObereEndlage$'failed at $'OUT_Endlage$'');
TEST_FINISHED();
```

maximbendiks, 2019-10-16:
Ah ok. I thought it was your test method.
If I have unterstood you correctly, you will need to call fb_DreieckTest more than once. For example, if you want to test BA=10, you will need something like:
```
TEST('FahrenAufObereEndlage');

fb_DreieckTest(IN_Start:=TRUE,
	IN_ZykZeit:= 100,
	IN_v:= 10,
	OUT_Sollwert=>OUT_Sollwert_Result,
	OUT_Endlage=>OUT_Endlage_Result); // here BA transits to 10
fb_DreieckTest(IN_Start:=TRUE,
	IN_ZykZeit:= 100,
	IN_v:= 10,
	OUT_Sollwert=>OUT_Sollwert_Result,
	OUT_Endlage=>OUT_Endlage_Result); // here you execute code in BA=10

// if you only wanted to check for BA=10, you can now execute your asserts and finish the test
AssertEquals_REAL(Expected := OUT_Sollwert_Expected,
                              Actual := OUT_Sollwert_Result,
			      Delta := Delta_Value,
                              Message := 'Test $'FahrenAufObereEndlage$'failed at $'OUT_Sollwert$'');
			 
AssertEquals_BOOL(Expected := OUT_Endlage_Expected,
                                Actual := OUT_Endlage_Result,
                                Message := 'Test $'FahrenAufObereEndlage$'failed at $'OUT_Endlage$'');
TEST_FINISHED();
```
After that you can create a new instance of FB_DreieckTest and create a new test method with other values because the tests should not depend on each other and you would want to start clean each test.

maximbendiks, 2019-10-16:
Your Test looks fine to me.

kumaraswamygaviyappa, 2019-10-17:
I was testing few more modules, it's good method for 100s of data, what if data are in 1000s or may be more. In the above example, **ARRAY** is used to store 40 data. Could you suggest a method or technique for larger volume of data.

maximbendiks, 2019-10-18:
I do not know your context, but it seems to me that you are misusing the concept of unit test. The idea behind unit testing is not to test as much data as possible but to find tests that cover all/most of your code. If you are interested, you can have a look at: http://blog.wingman-sw.com/tdd-guided-by-zombies
It always depends on the code you want to test, but for example it makes no sense to me to check this code ```
IF(OUT_Sollwert < (Endlage_oben - (IN_v * IN_ZykZeit)))THEN
	OUT_Sollwert:= OUT_Sollwert + (IN_v * IN_ZykZeit);
ELSE
```
20 times. For me as a reader it will make it harder to use your tests as a documentation. But thats only my opinion.

As for the storage of test data: Hard coding your data into an array is probably the best/safest/idealistic approach for unit tests. Write a script to convert your data from *.csv, database or whatever into simple TwinCAT code to initialize your arrays. Or read straight from your file or database.

sagatowski, 2019-10-20:
I just wanted to chime in and say that I agree with everything @maximbendiks has said.
 @kumaraswamygaviyappa, can I close this ticket?

kumaraswamygaviyappa, 2019-10-21:
@maximbendiks thank you for all your inputs.

closed 2019-10-21 by kumaraswamygaviyappa