tcunit/TcUnit #57: File Read for unit testing

kumaraswamygaviyappa, 2019-10-24:
I was able to read the **.csv file**, loaded Arrays  **Input_v_Data** and **OUT_Sollwert_Expected_Data** during Simulation.
when i am unit testing the **Function_Block** using **TcUnit**, i think, data is not read from csv file and Input_v_Data, OUT_Sollwert_Expected_Data always 0.

Question is: When unit testing using TcUnit, will files be read?

```
fb_TextModeRead(
	bRead:= TRUE,
	sFileName:=sFilePath,
	database=>OUT_database,
	bFileReadCompleted=>bIsFileReadCompleted);

fb_DreieckTest(
                IN_Start:=,
		IN_ZykZeit:= Input_ZykZeit,
		IN_v:= Input_v_Data[nIndex],
		OUT_Sollwert=>OUT_Sollwert_Result,
		OUT_Endlage=>OUT_Endlage_Result);		
		
AssertEquals_REAL(
               Expected := OUT_Sollwert_Expected_Data[nIndex],
	       Actual := OUT_Sollwert_Result,
	       Delta := Delta_Value,
	       Message := 'Test $'failed at $'OUT_Sollwert$'');
```
Line its executes inside fb_DreieckTest is :
`OUT_Sollwert:= OUT_Sollwert + (IN_v * IN_ZykZeit);`

Test Result
```
FAILED TEST 'PRG_TEST.fb_DreieckTest_Test@Test', EXP: 0.0, ACT: 6.0, MSG: Test 'failed at 'OUT_Sollwert' 
FAILED TEST 'PRG_TEST.fb_DreieckTest_Test@Test', EXP: 0.0, ACT: 7.0, MSG: Test 'failed at 'OUT_Sollwert' 
```

maximbendiks, 2019-10-24:
You have to wait multiple cycles until bIsFileReadCompleted is TRUE. Put your FB_DreieckTest and asserts into an if statement.

kumaraswamygaviyappa, 2019-10-25:
@maximbendiks Thank you very much for helping me again. It worked.

closed 2019-10-25 by kumaraswamygaviyappa

sagatowski, 2019-10-26:
@ascaron37 thanks for taking your time to provide great answers!