tcunit/TcUnit #60: The verifier should be integrated with TcUnit in one repository.

KubaO, 2019-12-06:
The tests aren't some foreign code: whenever you develop TcUnit, you should also be updating the tests. This is very hard to do when they are in split repositories. There's a requirement that the contributors add tests: well, TcUnit has no tests. They are some separate entity, and it's impossible to tell what particular TcUnit version they are for.

sagatowski, 2019-12-11:
There actually is a reference between them, the version in the project. Both of these are updated for each release, which I make sure to do just prior to releasing them.
That being said, I agree with the suggestion would make the maintenance easier not having them spread across two repos. In the original plan there was only the framework, the tester was added on later when functionality grew.
The PR #64 moves the tests into each separate directory of the root of TcUnit and updates the README.md. It would make more sense to keep both (tcunit and verifier) README.MD intact, and move the repo to github.com\TcUnit\TcUnit\TcUnit-Verifier (and in here the complete content of github.com\TcUnit\TcUnit-Verifier would go). I'll make sure to update the contributing.md to reflect the new location.

sagatowski, 2019-12-12:
The more I think about this the more it makes sense.
I've done some updates to TcUnit (and TcUnit-Verifier). Could you update the PR accordingly and put the verifier in the suggested location?

sagatowski, 2019-12-31:
Verifier integrated into TcUnit in commit b2b7a4235833fe1a63978ea03461a00a81f25d3e.
Thanks for this suggestion!

sagatowski, 2019-12-31:
Also, all issues (open & closed) from TcUnit-Verifier were moved to this respo.

closed 2019-12-31 by sagatowski