tcunit/TcUnit #69: Releasing openxml and .library sourcecode for future master release

Aliazzzz, 2019-12-22:
Dear TcUnit,

I would like to request that the code from the nextmaster will be made available as;

.Library,
OpenXml export,
.compiled-library,
And offcourse 
TcUnit sourcecode from the above artefacts are derived.

Reason is that I can use the .library (NOT to be confused with the .compiled-library) and the open xml exports to update CfUnit.

Aliazzzz, 2019-12-22:
Offcourse I mean all future master releases...

sagatowski, 2020-01-03:
For the very specific case of CfUnit using this, I think it makes more sense for the developers to simply download the code in a release (which is also provided for a release), and do the exports.

I think > 99% of the users of TcUnit only want to use the compiled-library (+ the documentation obviously). I'm worried that new people to the framework might get nervous when there are several files to download, while they only need one single of them.

Aliazzzz, 2020-01-03:
No problem, You are probably right that the majority doesn't need it. Anyway thanks for the consideration.

closed 2020-01-03 by Aliazzzz