tcunit/TcUnit #70: Implement optional parameters for the test results

sagatowski, 2019-12-27:
On top of #68, also implement the following optional parameters in the test results:
- Whether a test is skipped
- Number of assertions for a testcase
- failure message/type for a testcase
- Something more?

sagatowski, 2020-01-01:
As per request of a TcUnit-user, the information of whether a test has failed will be included as well.
[image]

Aliazzzz, 2020-01-01:
Okay,

1 thing:

Can you also send the code where you copy the info into the struct?

I’ll integrate the information in the jUnit output, I will concat it into the assert message in a neat way.

Plus I reallocated storedgeneraltestresults into the root of to the ST_TestSuiteResults.

[image]<https://user-images.githubusercontent.com/33381702/71646285-21d61580-2ce6-11ea-8533-2e7ce735f1e0.png>

sagatowski, 2020-01-01:
the result should not be in the st_testsuiteresults, as we only store the results there. The status-data is in the FB, as the variable: StoredTestSuiteResults + StoredGeneralTestResults. I can add a getter-method for these so that they are available.

Aliazzzz, 2020-01-01:
A getter method will solve the issue.
When will you commit your updates?

Aliazzzz, 2020-01-01:
A Getter will solve it cleanly yes.

sagatowski, 2020-01-01:
When it's done ;-)

sagatowski, 2020-01-01:
I don't think it's necessary with these getters, it will be redundant. You already got the information you need by first checking:
GVL_TcUnit.NumberOfInitializedTestSuites
And then for every test suite:
GVL_TcUnit.TestSuiteAddresses[TestSuiteNumber]^.AreAllTestsFinished()

If this is true, you know all tests are stored.
Look into:
FB_AdsTestResultLogger.LogTestSuiteResults()

of how this is utilized.

Aliazzzz, 2020-01-01:
No, I really do not agree on that last suggestion. In my opinion that cure is worse then the disease.
The answer should be publicly published, as it is a bad practice to calculate the same answers in multiple places, with all troubles in future code updates.
Plus, a ‘getter’ will not solve the public publish question as my FB consumes only ITF_TestSuiteResults. The getter is not inside that Interface so the third party FB cannot read the getter.
Looking forward to other suggestions.

Aliazzzz, 2020-01-01:
Suggestion!

I'll pass the instance of FB_TestResults as a pointer into my FB.
Then I can access its method and properties without problems.

sagatowski, 2020-01-02:
> No, I really do not agree on that last suggestion. In my opinion that cure is worse then the disease. The answer should be publicly published, as it is a bad practice to calculate the same answers [...]

You're right, it was a bad idea.
I would create an interface, such as I_TestResults, implement the method and do a dependency injection of that into all FBs that are dependent on it.

I'm not using properties in TcUnit so it would be nice to be consistent.

sagatowski, 2020-01-03:
I've commited the code with the interface now. Look at the AdsTestResultLogger of how it is dependent on the TestResults.

sagatowski, 2020-01-11:
Solved in 1f2da3a.
All optional parameters mentioned are implemented in the test results collector as well as included in the ADS test logger.

closed 2020-01-11 by sagatowski