tcunit/TcUnit #81: Checking wether testoutput is correct WITHOUT help of external resources?

Aliazzzz, 2019-12-31:
Hi,

The testsresults of the verifier are now checked with the help of an external resource in order to verify it's outcome on accuracy. It should in theory be possible to do this without an external resource. My suggestion is that we put some effort in trying to investigate this idea.

Aliazzz

Aliazzzz, 2020-01-03:
1) There is this CryptoLib to calculate hash value's
2) Maybe OSCAT can give us some functions.
3) The new FB_TestResults can be used for calculation of the hash.

sagatowski, 2020-01-04:
Where is the CryptoLib available? I could only find something for codesys (though not sure). Is it available for TwinCAT as well?

Aliazzzz, 2020-01-04:
CmpCrypto v3.5.15.0. for CODESYS. I highly doubt this is "rebadged" into TwinCAT by Beckhoff yet as they are at SP13. We should search the TwinCAT libraries for other usefull stuff. Plus we should search for an open source alternative to calculate a HASH code.

Aliazzzz, 2020-01-04:
"MD5_STR" from OSCAT NETWORK seems to fit;
```
Input 
   RUN: BOOL (Positive edge starts the calculation)
Output 
   DONE: BOOL (TRUE if calculations are complete)
   MD5: ARRAY[0..15] OF BYTE (current MD5 hash)
I / O 
   STR: STRING(string_length) (Text for HASH creation)

Example:
the MD5 hash of 'OSCAT' is 30f33ddb9f17df7219e1acdea3386743
```

We should only agree upon a few things:
What information are we going to feed it? 
   My suggestion would be to feed in every testresult only (succes, fail or skipped). 
Plus how is this information string going to be composed (nomenclature of the string)?
Suggestions?

sagatowski, 2020-01-08:
I don't have any opinions on the technical solution as long as it fulfills the requirements I have mentioned (the ones fulfilled by the .NET Verifier).

closed 2020-12-17 by sagatowski