tcunit/TcUnit #83: Add tests for WRITE_PROTECTED_ functions

sagatowski, 2019-12-31:
Add tests for the WRITE_PROTECTED_ functions in the .NET-verifier. There are already tests in the TwinCAT verifier, we just need to make sure these output (or rather, don't output) the correct values.

Aliazzzz, 2019-12-31:
The test FB for these tests is not fashioned in a real world application type of way. I will do a suggestion for an improvement on this test.

sagatowski, 2019-12-31:
Thanks for implementing this. Please add the class FB_WriteProtectedFunctions in the C#/.NET verifier and update the Program-class.

sagatowski, 2020-01-08:
Have you made any progress on this? Otherwise I'll implement it.

sagatowski, 2020-01-08:
Sorry, didn't have the patience to wait. Fixed :-)
Solved in commit c50dcd8.

closed 2020-01-08 by sagatowski