tcunit/TcUnit #85: LineIDs necessary?

sagatowski, 2020-01-03:
Everytime a file is changed, the TwinCAT IDE changes the contents of the file by adding the different lines such as:
<LineId Id="127" Count="0" />

to the POUs. Apparently there is a way to make TwinCAT ignore this, and put all of this into a separate file (which doesn't need to be version controlled).
This is described here:
https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_userinterface/18014403202147467.html&id=

This would however require:
1. That any contributor set his/hers IDE "Separate LineIDs = TRUE"
2. A manual job to remove all existing LineIds in the code

sagatowski, 2020-01-11:
Solved in commit 1f2da3a.
I've removed all LineIDs in all code which makes it much cleaner. Tested that everything still worked fine which it did. So far I've included the LineIDs.dbg-file (both for TcUnit and verifier), but it might be the case that these files can be ignored.

Also updated the CONTRIBUTORS.md file to explain that any contributor needs to update his/hers IDE so no new line ids will be introduced.

closed 2020-01-11 by sagatowski

Roald87, 2020-05-22:
Hi Jakob,

Can you share what you discussed with the Beckhoff support about the LineIDs? Why are these line ids there and why can they be ignored for the TcUnit project. Furthermore can they always be safely ignored? The InfoSys entry is not that informative about it.

Roald.

sagatowski, 2020-05-23:
Hi!

I had the conversation over phone, and I can't remember it exactly, but it was basically the information that is in the documentation:
https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_userinterface/18014403202147467.html&id=

The conclusion was that these could always be ignored, if you had the LineIds-file if you were going to handle breakpoints (or if it was just doing anything "online"/debugging in general).