tcunit/TcUnit #89: Test with Counter in the same FB

YokMenns, 2020-01-16:
Hi sagatowski ,
i convert your actual tcUnit 1.1.0.0 in libray and i try to write my tests with it.
It looks like if i have a counter in the FB and if i define a counter in the FB_Test ( refere to your example with the timer), the system don't show me any testresult more. I mean in the lg i see the failed tests but in errors and the other tests as messages but the test about checking the counter still missed on the log and also the rsult statistic still missed. i know the tcUnit isn't release, but maybe it is better to know it now. Or maybe i make something false. Please can you check it. Thank you

YokMenns, 2020-01-16:
I don't think it is a new issue. Because it is abut a non release tcUnit. So i put it as comment

closed 2020-01-16 by YokMenns

sagatowski, 2020-01-17:
Duplicate. Issue resolved in #20