tcunit/TcUnit #92: Duplicate event message output

DavidHopkinsFbr, 2020-01-29:
For some reason I'm getting three copies of each output event:

[image]

Any suggestions where I should look to start debugging this? I'm running TcUnit 1.0.0.

It might be relevant that the structure of this particular solution is like this:

- TwinCAT Project 1
  - PLC Project 1
- TwinCAT Project 2
  - PLC Project 2 (currently disabled)
  - C++ Project
- TwinCAT Project 3
  - PLC Project 3
  - PLC Project 4 (this is the one which contains the TcUnit code)

Project 3 is the one which I'm actually activating the configuration of.

I'm speculating that the multiple TC projects in one solution might be related to the duplicated event messages, though I'm not sure what the mechanism for that might be.

DavidHopkinsFbr, 2020-01-29:
Each failed assertion is also output 3 times (when there are any)

DavidHopkinsFbr, 2020-01-29:
A bit more testing shows that I get one duplicate output message for each top-level TwinCAT project in the solution. If I unload the other projects before activating the configuration, I only get one copy of the output messages.

DavidHopkinsFbr, 2020-01-29:
I've made a simple demo solution you can play with here: https://github.com/DavidHopkinsFbr/TcUnitDuplicateMessageOutput

sagatowski, 2020-01-29:
Thanks for this very well documented issue. I'll look into it during the week.

sagatowski, 2020-01-29:
A quick test indeed seems to show that this is related to the multiple TwinCAT projects and not multiple PLC projects.
I've tested to remove the extra PLC project, and there was no difference although if I remove the extra TwinCAT project I only got one single output.
This is very interesting considering that only one of the TwinCAT projects in the example is running (as both are on port 851).

I never considered the case of multiple TwinCAT projects in one solution. What is the use case for this? How does it differ from having multiple PLC projects in one TwinCAT project?

Issue is the same in the current development branch (1.1-working) of TcUnit.

Aliazzzz, 2020-01-29:
This seems a TwinCAT / VS.net issue;

AFAIK CODESYS does not support "solution files" or "multi projects" in a single project container.
CODESYS only supports multiple PLC projects within a single single project container, e.g;

CODESYS Project
     PLC1 Project
     PLC2 Project
     etc.

DavidHopkinsFbr, 2020-01-30:
In this case, there are three projects in the solution for [hysterical reasons](http://www.catb.org/jargon/html/H/hysterical-reasons.html). This is the result of combining several different projects (TcCOM module, plus a PLC library wrapper for the TcCOM interface, plus a test deployment) into a single solution. I'll probably reorganise them into a single TwinCAT project with multiple PLCs at some point.

An _actually good_ use case for multiple TC projects in one solution is when you have multiple separate TwinCAT runtimes that are closely coupled, e.g. via EtherCAT Automation Protocol (EAP) PLC-to-PLC variable mapping, and you want to keep them together in source control & VS to simplify change management across that interface. For example, you might have an architecture where you have a high-level supervisory PLC which coordinates the activities of several smaller PLCs that each manage one module of a production line or machine. It can be helpful to be able to manage all that code through one Git repo, have one feature branch and pull request covering a change that affects multiple nodes, and all the code accessible in a single VS instance.

That leads me on to a theory of what's happening in this test case. In my solution, and the test solution, all the TwinCAT projects _are targeting the same runtime_. I think TwinCAT XAE is subscribing each of those projects to the TC event logger on that project's target runtime. And, since all the projects are targeting the _same_ runtime, you get multiple subscriptions to the same event logger. That would naturally lead to XAE displaying the same message multiple times.

So on reflection, I think this might not be a TcUnit bug. If I'm right about the cause, it's more an undesirable artefact of a mis-structured solution, since you wouldn't _normally_ have multiple TC projects _targeting the same runtime_.

I think the issue is that all the TC projects in the solution share the same target: the local runtime.

sagatowski, 2020-02-02:
David, first thanks of the analysis and excellent description of the “good’s/bads” of multiple TcProjects in a single solution. This for sure shed some new light and gave me some new insight into this matter.

I’ve looked more deep into this and I have concluded the same thing as you. You actually don’t need to even run TcUnit to see this behaviour. Using your example, if you look into the error-list of Visual Studio you can see that you get a double of the other events as well:
[image]

Just to be on the safe side, I added a counter to TcUnit everytime I called Tc2_System.ADSLOGSTR(), and indeed the counter was half of the events compared to what I could see in the eventlogger, which means that the call to Tc2_System.ADSLOGSTR() happens the right amount of times (using TcUnit v1.1 it does 14 calls to Tc2_System.ADSLOGSTR() but I got 28 event logs).

I’ve sent an e-mail to the Beckhoff support, and see if we can get any more information, and more importantly, if there is anything we can do for a work-around in this matter.

I’m not going to mark this issue as a bug in TcUnit, but as “question”, and see what we get from Beckhoff.

sagatowski, 2020-03-20:
I still haven't been able to get any information from Beckhoff regarding this. Maybe someone else has some input to provide?

sagatowski, 2020-09-03:
I just wanted to say that I finally got a response from Beckhoff and they have confirmed that they will fix this issue in a future release (no information on when though). I'll close this ticket.

closed 2020-09-03 by sagatowski

Roald87, 2021-01-26:
A work around could be to temporarily unload the projects you do not need.

[image]

sagatowski, 2021-02-24:
Another update. Beckhoff will NOT fix this, as this behaviour is "per design".