tcunit/TcUnit #93: Linking FB from development project to Testing Project

kumaraswamygaviyappa, 2020-02-05:
I have created a **testing project** to run unit testing from **TCunit framework** and wanted to link files from **development project** but there is only Add option when i right click on POU folder in solution explorer.
When i did so, a copy is made in testing project POU folder.
The problem i am facing is, whenever there is change in the development project files i need to update POU folder in the testing project.
Please let me know if there any solution.

sagatowski, 2020-02-06:
I'm not entirely sure I understand the problem. Do I understand it correctly that you have one project where the actual (non-testing) code is (let's call it A), and then you have another testing project (let's call it B), and in B you import the POUs from A?

Why don't you add the unit tests in A?

If you for some reason need them strictly separated, you can export A as a library, and use A as a library reference in B (although this adds administrate work as the combination of the two projects won't be atomic and you will need to handle that)

kumaraswamygaviyappa, 2020-02-12:
Yes you're right @sagatowski , I want to test POU's of project A in project B. 
The reason for making two different projects (A/Development and B/Test) is, I would like to carry out testing of the developed POU’s in the separate test folder without disturbing the development process. Is this a bad idea?

I would like to ask, why the referencing is not provided in twinCAT. I feel it important feature and also present in other language and IDE’s .

sagatowski, 2020-02-24:
The tests should not be seen as a separate entity to the software, but rather as a vital part of it, but instead of going into the theoretical discussions of this, if you want to have them as separate projects one option is to simply export the main project into a library (*.compiled-library or *.library). You simply export and install the main project as a library in your developer machine.

When you export the main project as a library, you can add a reference to it in the test project. Now you have access to all the POUs in the test project.

A good resource for this is:
https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_plc_intro/36028801208102411.html&id=172749180532672501

The problem with this is that it adds complexity and makes maintenance more cumbersome, as it adds more work to make sure the two are synchronized.

This might work for one or two projects, but once you have a big enough project (or several projects), I'd recommend keeping these two in one solution in all other cases.

kumaraswamygaviyappa, 2020-02-25:
@sagatowski Thanks for the input.

closed 2020-02-25 by kumaraswamygaviyappa

u-hafner, 2020-02-25:
Just as comment for info: My usecase would also be to separate testing code from production code. Especially because we deliver code to clients and don't want to hand out the testing code.
I already tried out to use it as a library but wasn't able to automate it yet as the only option I know to create a library is to right click on the plc part and manually create a new version of the library.
But I will keep an eye on your discussions to improve my workflow :-).

sagatowski, 2020-02-26:
@u-hafner If you're willing to put some time and effort into it, you could partially automate your process by having a build-server doing the creation of the library for you on your machine (after every save, or after commit to VCS for instance) by using the TwinCAT automation interface. The creation of libraries from a PLC project is included in the automation interface.

u-hafner, 2020-02-26:
yes, thanks. That's about the information we had with automation interface. Haven't had the time yet but we are currently preparing exactly that to hopefully automate our process.
The final goal would be to automate from checkin to test and then checking test reports for failed tests to verify the checkin. (I know this is not a discussion forum so I'll keep it short.)

kumaraswamygaviyappa, 2020-03-06:
It worked using [symbolic-links](https://www.howtogeek.com/howto/16226/complete-guide-to-symbolic-links-symlinks-on-windows-or-linux/). Received one more suggestion from [Here](https://stackoverflow.com/questions/60071536/linking-functional-block-fb-from-development-project-to-testing-project-twinc?answertab=oldest#tab-top)