tcunit/TcUnit #94: TcUnit Does not work anymore after infinite loop in testmethod

larshaarmeijer, 2020-02-28:
I was building a test to Test Method a function block with CASEs. To be able to check how the function block loops through the cases, it was possible to enter the FB multiple times. For a certain CASE, i did not know how often it needed to enter that FB, so i created a loop, which could loop until the Result equals the ExpectedResult. However, i did not insert a maximum amount of loops to prefent the infinite loop (yes, very stupid...). 

The thing is, after the program ended in that infinite loop, and i just restarted the TwinCAT Shell, the TcUnit does not work anymore (it doesn't give me the results anymore in the error list, see screenshot attached). I've tried to create complete new projects, but even in those it does not work anymore.

Do you have some suggestions for me to fix this, i really like the TcUnit and would like to use it further...

[image]

sagatowski, 2020-02-28:
Hi @larshaarmeijer ! I'm very happy that you find TcUnit useful. I'm not entirely sure I understand the problem correctly. What do you mean that the program ended in the infinite loop? Could you upload your project, so I can see that I can reproduce the problem where the logs are not printed?

If nothing is printed to the error-list (not just for TcUnit), it might be that the ADS-messages (which TCUnit is using to print to the error-logger) are not working, which might be because of the AMS/ADS router. Have you tried rebooting the machine?

Also, generally if you want to assert only once a certain criteria has happened, I would suggest you to add a timer (TON) in that test so that the test doesn't run indefinitely. What you can do is then to simply assert the result (and do a test_finished()) once the timer has elapsed.

larshaarmeijer, 2020-02-28:
Hi @sagatowski, thanks for the quick reply! I'm sorry for the unclearness, so let me explain more about where it went wrong:

Take this example of the case:
[image]

Very simple of course, but in my former project I could not know when it ended. So i created the following loop in the Test Method:

[image]

Since a ran the TcUnit on the former project (similar to this one) i've never seen the test results again. 

To prevent this, i planned to build something like this:
[image]

I've already rebooted the system, reinstalled TcUnit, started a completely new project... 

If I make a typo in the code, it does show up in the Error list. So that makes me think the AMS/ADS Router is not the problem...

I hope this is enough information :)

sagatowski, 2020-02-28:
@larshaarmeijer , thank you for your excellent problem description.
The problem is that you in your test is always calling the FB_Case with CaseState := 0.
`CaseBlock(CaseState := 0, CaseCompleted=>Result`
This means that the CaseState inside FB_Case will always be overwritten to 0, and thus it will never reach 50 (and thus CaseCompleted = True).

larshaarmeijer, 2020-02-28:
Oke very much thanks! I'm using it for work, so I won't look into it untill monday... take it easy:)

larshaarmeijer, 2020-02-28:
> You are sure you are calling the IfCase0True()-method from the test-suite? Also, your testsuite extends the FB_TestSuite function block? And also, you are initializing the test-suite in your PRG_TEST/MAIN?

Yes, i've even downloaded your SimpleMath solution again, even that one didn't work anymore

sagatowski, 2020-02-28:
@larshaarmeijer Ok, that sounds very strange indeed. And you have TcUnit 1.0 (release) installed?
https://github.com/tcunit/TcUnit/releases

larshaarmeijer, 2020-02-28:
TcXaeShell
Version 15.0.28010.2050 D15.8
Microsoft .NET Framework
Version 4.8.03761

Installed Version: Express

AMLImport Extension   1.0
AMLImport Visual Studio Extension Detailed Info

ASP.NET and Web Tools 2017   15.8.05085.0
ASP.NET and Web Tools 2017

BodePackage Extension   1.0
BodePackage Visual Studio Extension Detailed Info

FilterPackage Extension   1.0
FilterPackage Visual Studio Extension Detailed Info

GitAdsServiceSettingsToolWindow Extension   1.0
GitAdsServiceSettingsToolWindow Visual Studio Extension Detailed Info

JavaScript Language Service   2.0
JavaScript Language Service

NuGet Package Manager   4.6.0
NuGet Package Manager in Visual Studio. For more information about NuGet, visit http://docs.nuget.org/.

ProjectServicesPackage Extension   1.0
ProjectServicesPackage Visual Studio Extension Detailed Info

TcProjectCompare   1.0
Visual Studio Integrated Version of TcProjectCompare

TcTargetBrowserPackage Extension   1.0
TcTargetBrowserPackage Visual Studio Extension Detailed Info

TcXaeDebuggerLiveWatch   1.0
TwinCAT XAE Live Watch Tool Window

TcXaeHelper   1.0
Info

TcXaeModules   1.0
TwinCAT TMC Editor, TwinCAT TMC Code Generator

TcXaeShell   1.5.0
TwinCAT XAE Shell customization

TcXaeStartPage   1.12
Beckhoff Start Page for TwinCAT XAE

TwinCAT Measurement   3.4.3142.4
This package contains the project system for TwinCAT Measurement projects.

TwinCAT Scope   1.0
Integration package for TwinCAT Scope components

TwinCAT XAE Base   3.1.0.0
The TwinCAT XAE Base extension provides the basic project structure for TwinCAT projects.

TwinCAT XAE EventLogger   1.0
The TwinCAT XAE Eventlogger extension provides the capability of monitoring events sent through the TwinCAT 3 Eventlogger of a target system.

TwinCAT XAE PLC   3.1.0.0
PLC Environment

larshaarmeijer, 2020-02-28:
I'm coming from Mechanical Engineering and I'm only into PLC/Software for a year, so it could also be my minor understanding of software haha...

larshaarmeijer, 2020-02-28:
> @larshaarmeijer Ok, that sounds very strange indeed. And you have TcUnit 1.0 (release) installed?
> https://github.com/tcunit/TcUnit/releases
> Also, which version of TwinCAT/Visual Studio are you running?

Strange thing with this is that i've succesfully ran the TcUnit multiple times before the fialure occured...

sagatowski, 2020-02-28:
I just tested TcUnit with the simple math project and the same version of your software and it worked all fine. Are you sure you are:
- Activating the solution
- Starting the project?
- And that the project is actually running?

Just add a simple counter to the PRG_TEST of simple math project, increase it by 1 in every cycle, login to the PLC and see that it actually increasses.

I've seen people having the "autostart"-flag enabled and just assuming it will always be like this. But if you play around with TwinCAT a lot this can sometimes dissapear and you need to either add the autostart flag again or manually start the project...

larshaarmeijer, 2020-02-28:
Now i logged in, and started the project, it worked.

earlier only activating the solution already worked, don't know why that's not the case know. 

Most importantly, i can continue...

Many Thanks for your time, and the building of this whole program.

sagatowski, 2020-03-01:
@larshaarmeijer fantastic! You probably had the autostart boot project flag enabled for that particular project, and for some reason it might have been disabled/removed:
https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_plc_intro/2531433611.html&id=

Closing this ticket.

closed 2020-03-01 by sagatowski