tcunit/TcUnit #99: Multiple Suites fail to complete

RogerChristopher, 2020-04-14:
If I have two test suites that take multiple scans to complete, the test run ends when the first test suite has completed (the second never finishes as the test runner thinks all suites have completed)

sagatowski, 2020-04-14:
Could you provide an example project where this is reproduced?

RogerChristopher, 2020-04-14:
Two test suites in the attached file. Run the test runner with these and the tests will not complete.
Pull request submitted which changes NumberOfTestSuitesFinished from VAR_INST to VAR, which fixes this problem. (Tests complete correctly after this mod).

[MultipleTestSuitesError.zip](https://github.com/tcunit/TcUnit/files/4475822/MultipleTestSuitesError.zip)

sagatowski, 2020-04-14:
@RogerChristopher , thank you very much.
I'll look into it during the week.

sagatowski, 2020-04-14:
@RogerChristopher Quickly opened the file, but it was just an XML-file. Is it possible for you to upload a complete TwinCAT project where the problem is reproduced?

RogerChristopher, 2020-04-14:
It is a PLC-Open export file containing the two test suites, each with one test. One takes 1 second to complete, the other takes 2 seconds to complete. Try importing it into a project and add these to a test runner - should show the issue.

RogerChristopher, 2020-04-14:
Or try the attached project...

[TwinCAT Project7.zip](https://github.com/tcunit/TcUnit/files/4476266/TwinCAT.Project7.zip)

sagatowski, 2020-04-14:
Bug fixed in PR #100

closed 2020-04-14 by sagatowski