tcunit/TcUnit #102: Tests with timers and multible TestSuites

kaelev7, 2020-05-13:
Hi,   
I am writing tests for a library with watchdog functions. For the watchdogs I am using timers.  The tests for timer are written according to examples given in issue #20. For the first test suite everything is fine and works as expected. But as soon as use a second FB that extends *TcUnit.FB_TestSuite* the first test that has a timer inside will never be finished. If I watch in the online view I see that the timer gets started but only gets called twice (ET is T#20ms, with a cycle time of 10ms).   
As soon as I only use one test suite again it works.  (Just comment all out and using only one).  
   

I'm using TcUnit 1.1.0 TwinCAT v3.1.4024.7 and Visual Studio Community 2017 Version 15.9.21

RogerChristopher, 2020-05-13:
I think this is issue #99, fixed with PR#100. Try the latest version
(though it doesn't look like the version number was incremented after
PR#100)
Roger Christopher

sagatowski, 2020-05-13:
Please check latest version from code (non-release) as @RogerChristopher.
The reason the version number was not incremented is that because version 1.1 is still not released (latest formal release is 1.0).

kaelev7, 2020-05-13:
Thanks for the fast response.

With Version 1.0 its works.
To get the last version of master branch to work I had to remove TcUnit completely, even delete the library within the TwinCat folders, restart the system and install it again.

closed 2020-05-13 by kaelev7

sagatowski, 2020-05-13:
@kaelev7 Thanks for the feedback.
Version 1.1 should hopefully be released in the coming weeks.

Aliazzzz, 2020-05-13:
> Version 1.1 should hopefully be released in the coming weeks.

Yes! If you do I can release the CODESYS version of it also!

kaelev7, 2020-05-13:
@sagatowski  no need to hurry. Both version working for me.