tcunit/TcUnit #106: implementing WSTRING support

Aliazzzz, 2020-05-16:
Dear Jakob,

In the pure CODESYS [CfUnit-fork](https://forge.codesys.com/svn/prj,cfunit,code/landingpage/index.html?raw) of TcUnit my supporters ask for WSTRING support by popular demand(!).

Supporting this datatype will open up testing www/ http/IIoT targeted applications of which many implement unicode encoding instead of ASCII encoding.

Suggested Datatype:  
`
TYPE T_MaxWString : Alias WSTRING(1024);  // 1024 is an arbitrairily chosen maximum
`

Suggested Assertion Prototype: 
`
AssertEquals_WSTRING( Expected : T_MaxWString, 
                                Actual : T_MaxWString, 
                                Message : T_MaxString);
`

For WSTRING datatype see: https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_plc_intro/27021600293660299.html&id=

I hope you will take support for this datattype in (future) consideration.

With kind regards,

Aliazzzz

Aliazzzz, 2020-05-16:
PS

The AssertAny also does not implement WSTRING at this moment...

sagatowski, 2020-05-18:
Great suggestion! I think I'll stick to using the standard of 255 characters as this is what Beckhoff is using in all their WSTRING-functions (just as with STRING).

AssertAny does actually support WString indirectly, as it compares the two objects by a byte-by-byte comparison if it doesn't find a match on the supported type.

sagatowski, 2020-05-25:
Solved in commit 13276ed.

closed 2020-05-25 by sagatowski