tcunit/TcUnit #110: Implement testsuites as FB with linked lists and a test as a linked list element

Aliazzzz, 2020-05-24:
On the topic of improvements/redesign ideas;

I was wondering if a TestSuite could be implemented as an FB with several linked lists and a test could be implemented as a single linked list element. The advantages of a linked list implementation over array's is a significant gain of flexibility/scalability and the skipping of unnecessary extra household actions. 
The testsuite could implement a 'queue -list' , a 'skipped -list', a 'failed -list' and a 'succes -list'.
All the tests from the queue should then be run sequentially and then moved according of the result to a target-lists ('skipped -list', a 'failed -list' and a 'succes -list'). The end result is then automatically stored in the target lists and do not need apart from tallying the totals. Thus, every testsuite is now automatically self-describing and can be printed or accessed on-demand.

Offcourse such advantages have a penalty of extra overhead (i.e. the list iterators). Also, an array is probably faster in processing, but this only a trivial issue with computer power these days. Anyway, a linked lists of tests will only contain only the instantiated tests, no more, no less. This will reduce the reserved memory in the runtime considerably in runtime compared to static arrays. My feeeling says the implementation of linked lists will be favorable over array's when the number of testsuites and tests grow. Surely, there will also be a break evenpoint somewhere but sooner or later the linked list implementation will prevail (not tested, but based on gut)

An example of skipping a test would be to store a test set to skipped in a special list for skipped elements. Another advantage is the iteration through the elementslists is by design simpler.

Anyway, a linked list can also offer smarter skipping of tests and other not yet considered bonuses.

Anyway, I'd like to start a discussion on this point...

Aliazzz

Roald87, 2020-05-25:
Beckhoff has an [example](https://infosys.beckhoff.com/content/1033/tcplclib_tc2_utilities/18014398544900235.html?id=4803292355478825527) of how a linked list can be implemented. I've used this in a project before and works quite well if you want to have some kind of queue system.  Disadvantage is that the concept is a bit harder to grasp and debug than an array which contains all your tests.

Aliazzzz, 2020-05-25:
I use CODESYS the 'vanilla' version of TwinCAT. Most of the time and with some experience, the examples are interchangeable between TwinCAT and CODESYS.

Funny that you mention the debugging of linked lists ;-). As per CODESYS 3.5 SP16 (just got released), the debugging of linked lists has been eased significantly(!) as the debugger now shows you all relative info of the linked list and its elements while before it did not. I expect this feature to be introduced downstream into TwinCAT in due time (dont ask me when... I dont know).

RogerChristopher, 2020-05-25:
The Beckhoff linked list implementation has a fixed maximum length and uses an array of nodes internally, so saves nothing over an array. Its harder to add/delete tests because the API is horrible (in my opinion).

The idea is sound, though, but perhaps not worth the increase in complexity.

Aliazzzz, 2020-05-25:
The CODESYS implementation implements FB's with at least two adresses (.Prev / next element). It also implements a simple register/deregister method to add it to a list. Also, in no way does their solution use an internal array of any kind. That would make no sense as the component should be (apart from the adressing) self managed and self contained. So it is a pure linked list..
Also they don't have a limitation on components, it is infinite ( or as far as the platform resources stretch ;-) )
The API (or List iterator and component FB's implementation and documentation) is well documented and field proven. Plus your feedback makes me wonder why Beckhoff has chosen to supply their own (bespoke) solution for this?

Sure, complexity increases in housholding of the list but in the end this is not a big issue since machines have memory and CPU power enough. I doubt if the extra resources outweigh over the gained flexibility.

sagatowski, 2020-05-25:
I have to agree with @RogerChristopher here. I like the suggested concept, but considering the implementation of linked lists in TwinCAT there really is no gain here. I've used the LinkedLists in TwinCAT and the API is terrible. It's using actions back and forth and makes the code clutterly and ugly.

Does this mean that the codesys linkedlists are allocating memory in runtime (dynamically)? AFAIK, there are no Beckhoff-libraries in TwinCAT that use the [__NEW operator](https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_plc_intro/72057596567099019.html&id=) in the background. In TwinCAT this allocates memory from the router memory that you can pre-allocate up to a certain size. Is it working similarly in codesys vanilla?

Aliazzzz, 2020-05-25:
_Does this mean that the codesys linkedlists are allocating memory in runtime (dynamically)? AFAIK, there are no Beckhoff-libraries in TwinCAT that use the __NEW operator in the background_

=> No, every test is instantiated during design-time. So the compiler knows beforehand what the memory usage will be. When you think about it, the Linked list component itself will not "move" through the memory. Its just a matter of keeping track of relations i.e. who points to who.

As a bonus the code does ** not ** utilise __NEW, but that option can be enabled if wished/permitted. When enabled the test framework design has a maximum flexibility and scalability since in running such a scenario, FB's can be created or destroyed under Runtime. Plus, as I guess the test framework program (aka the unit testing software project) itself usually does not run in a production environment but mostly in a lab setup, the deployment of the __NEW operator is less "dangerous" and can be enabled under consideration.

CODESYS documentation warns users to not use the __NEW operator in a production environment where stability/reliability is key... So using it in a production environment is at your own risk.

Aliazzzz, 2020-06-22:
Maybe we can try to implement these suggestions for v2?
Then again, v1.1 is long overdue...

sagatowski, 2020-06-23:
There is little to no gain with implementing this with the Beckhoff implementation of linked lists, I don't think it's worth the effort.

closed 2020-12-17 by sagatowski