tcunit/TcUnit #111: Clarification of contrib guidelines

I-Campbell, 2020-05-29:
Hello:

### [background]
I am looking to better contribute to this project, perhaps with some coding or some ideas.  I use cfUnit, a translation of TcUnit, and would like to be able to contribute bug fixes and improvements upstream.

### [Steps to reproduce]
Go to https://github.com/tcunit/TcUnit/blob/master/CONTRIBUTING.md

### [Notes]
###### 1. "... send an e-mail with the API change..."
Should this read "include the API change in the Pull Request"?  If not, to whom do I send such an email?

###### 2. "Tools→Options→TwinCAT→Write options→Separate LineIDs"
In 3.1.4022.32 this is Tools→Options→TwinCAT→**PLC Environment**→Write options→Separate LineIDs"

###### 3. "Visual Studio 2013 Community Edition." I would like to use a different toolset because...
According to https://visualstudio.microsoft.com/vs/compare/ if I am in a company that makes over USD1m/year, I cannot use TwinCAT with Visual Studio Community Edition for commercial purposes.  Sure, I can use it for contributing to TcUnit, as TcUnit is open source, but I couldn't then use that same environment to go and program a machine.
Would the project consider an additional configuration option? I haven't yet fully tested this works with the TcUnit project:
a. TwinCAT Version defined at TcUnit.tsproj + Latest VS shell that ships with that build (eg 4022.x + VS2013 Shell.
b. Latest Version of Visual Studio Code for the C# project, which is itself MIT licensed (https://github.com/microsoft/vscode/blob/master/LICENSE.txt)
c. from [there](https://dotnet.microsoft.com/download/dotnet-framework) download dotnet framework SDK version as defined here: [TCUnit-Verifier/App.config](https://github.com/tcunit/TcUnit/blob/13276eda29cdc6256313e8969f2a41e4826f3933/TcUnit-Verifier/TcUnit-Verifier_DotNet/TcUnit-Verifier/App.config#L4) (free to use if you have a windows OS license).  

Eventually we may migrate to dotnet core, which is itself open source and free to use (dotnet Framework Runtime on the other hand is "Free with every purchase of windows".)  But that is another issue for another day.

Note I haven't tested my alternative toolset.

If you let me know your thoughts on 1, 2 and 3, I would like to test and then make a pull request with the changes to contributing.md and steps for setting up the alternative programming environment.

sagatowski, 2020-05-30:
Hi @I-Campbell ,  and welcome! Always appreciated with new contributors!
About (1) and (2), you're 100% correct. Thanks for finding this.

Regarding (3), you're also right regarding the usage of community edition. However, I think the easiest approach to this is to create a separate virtual machine with VS2013 if you want to maintain TcUnit. This is most likely the best approach anyway, as you might already know that there are might be some problems loading an older version of TwinCAT if you have a newer one (for example, installing 3.1.4022.32 if you have 4024.7, then you have to go through the remote manager installation). Creating a virtual machine with a clean 4022.x-installation, you can use this environment only for TcUnit and not for anything else.

I think this is cleaner, and for a potential new contributor that works on a company with more than 1M-USD I don't want to scare them away with over-complicating the prerequisites for contribution :-)

I would however appreciate if you could look into whether it's possible to setup a configuration as you suggested, and this might be considered for the future.

/ J

I-Campbell, 2020-06-17:
Update: 
(3) Can confirm that the MIT licensed VSCode works great with compiling the C# verifier.  It uses the same compiler as visual studio, which you would specifically download from https://dotnet.microsoft.com/download/dotnet-framework/net461 (or whichever version is currently listed here:) https://github.com/tcunit/TcUnit/blob/4961aa9e036c6eae6a4a46422a37bd35f461c35c/TcUnit-Verifier/TcUnit-Verifier_DotNet/TcUnit-Verifier/TcUnit-Verifier.csproj#L12

sagatowski, 2020-06-23:
Solved in #115

closed 2020-06-23 by sagatowski