tcunit/TcUnit #113: Timed Testing in multiple testsuites

Aliazzzz, 2020-06-05:
Hi,

Could somebody post a small and working timed tests in multiple testsuites example?
Preferably a bit different then the last example that was posted here some year ago ...

Thank you kindly in advance and I hope to hear from you all ;-)

Roald87, 2020-06-09:
I'm not sure if I understand what you mean. Can you elaborate a bit? Do you mean how long the execution took for each test suite or all test suites combined? And do you mean including the activation time of the project? Also what is the purpose?

Aliazzzz, 2020-06-09:
question in short;
If you have testsuites that are time constrained or state dependent, would you like to post some unit test examples (ways of tackling such issues) ? Thank you kindly in advance!

purpose;
Usually a software project can contain one or several function block which have a state machine implemented or are in a way time constrained or even both. To test such a FB we can choose to implement a timed test. Such test software can be implemented in various ways, there is no right or wrong for it. Currently I have seen just a 1 (a single) approach for of a timed test. 
However, I'd like to see more approaches (different approaches) to timed/state tests so we can offer a variety of solutions to such usecases.

Roald87, 2020-06-09:
You mean how to solve #54?

Aliazzzz, 2020-06-09:
If you have testsuites that are time constrained or state dependent, would you like to post some unit test examples (ways of tackling such issues) ? Thank you kindly in advance!

sagatowski, 2020-06-09:
You have one example here:
https://tcunit.org/frequently-asked-questions/?Display_FAQ=703

I-Campbell, 2020-06-17:
VAR_INST for variables in methods that you want to keep static. builds on sagatowski's linked FAQ 703
https://infosys.beckhoff.com/content/1033/tc3_plc_intro/9007201783539083.html

Aliazzzz, 2020-06-18:
Thanks for the anwsers, the question is resolved.

closed 2020-06-18 by Aliazzzz