tcunit/TcUnit #117: AssertEquals(ANY) with type BIT compares the whole byte

I-Campbell, 2020-08-04:
If the BITS are in the same BYTE it will fail to detect an Assert Failure. (Ie. always asserts pass)
If the BITS are in different BYTES it will give an Assert Failure all the time, unless the other bits are identical.
Two incorrect asserts shown below (first one Fails, when it should succeed.  Second Succeeds when it should fail)

```
FUNCTION_BLOCK AssertEqualsANY_BIT EXTENDS tcUnit.FB_TestSuite
VAR
	OtherBit1 : BIT := FALSE;
	Expected : BIT;
	OtherBit2 : BIT := FALSE;
	Wall   : DINT;
	OtherBit3 : BIT := TRUE;
	Actual : BIT;
	OtherBit4 : BIT := TRUE;
	Wall2   : DINT;
	
	ExpectedFailure : BIT;
	ActualFailure : BIT;
END_VAR

tcUnit.TEST('AssertEqualsANY_BITEqual');
Expected := TRUE;
Actual := TRUE;

AssertEquals(Expected := Expected, 
                 Actual := Actual, 
                 Message := 'Asserting True is True falied');
tcUnit.TEST_FINISHED();

tcUnit.TEST('AssertEqualsANY_BITNotEqual');
ExpectedFailure := TRUE;
ActualFailure := FALSE;

AssertEquals(Expected := ExpectedFailure, 
                 Actual := ActualFailure, 
                 Message := 'Asserting TRUE is FALSE falied (as it should!)');
tcUnit.TEST_FINISHED();
```

sagatowski, 2020-08-05:
AssertEquals(Any) doesn't support the BIT-type (yet, I guess this should be possible). A Bit takes up a full byte in TwinCAT.
BIT can only be used in structures and function blocks, and not as inputs (that's why there is not AssertEquals_BIT).

sagatowski, 2020-08-07:
I tried to implement support for BIT but was just reminded that it's actually not possible. There is no way to have var_inputs to bits and thus with the current design of TcUnit it's not possible to add it.
One possibility is to simply use the BOOL-variant and do a TO_BOOL (or BIT_TO_BOOL) conversion prior to asserting.

sagatowski, 2020-08-07:
I can indeed confirm that a BIT_TO_BOOL-conversion works just fine. I have added an example to the F.A.Q. of TcUnit:
https://tcunit.org/frequently-asked-questions/?Display_FAQ=871

closed 2020-08-07 by sagatowski