tcunit/TcUnit #120: Page Fault when starting TcUnit.RUN() without any TestSuites installed

oswin02, 2020-09-04:
When I run only TcUnit.RUN() in an empty project without any TestSuites, the program stops in FB_TestResults at line 61 with this error message:
`Exception (Exception Code: 0xc0000005, Page Fault) in PLC Application .... Instance, Task PlcTask (RBP: 0xffffc4809ef4ec60, RIP: 0xffffb4865cf8bac4, RSP: 0xffffc4809ef4e5e0)`

sagatowski, 2020-09-04:
I have confirmed this is indeed an issue. Good catch @oswin02 ! Will be solved prior to release of version 1.1.

oswin02, 2020-09-04:
This here at the first lines of FB_TestResults seem to fix it, but it's just quick and dirty:
```
IF GVL_TcUnit.NumberOfInitializedTestSuites = 0 THEN
	RETURN;
END_IF
```

oswin02, 2020-09-04:
This subrange type could be the reason for the page fault: 
`StoringTestSuiteResultNumber : UINT(1..GVL_Param_TcUnit.MaxNumberOfTestSuites);`

sagatowski, 2020-09-05:
Solved. See #121 .

closed 2020-09-05 by sagatowski