tcunit/TcUnit #128: Create Error Message for Test > MaxNumberOfTestsForEachTestSuite

Beidendorfer, 2020-12-20:
If there are more tests in a TestSuit than are set in the  parameter list MaxNumberOfTestsForEachTestSuite (100)
All tests above this limit will be marked as failed, but no error log will be issued.

 It would be nice to have a log entry here that says "There are more tests than MaxNumberOfTestsForEachTestSuite, please change the value".

Example: No exact information why the tests failed 
(Reason: More Test(216) in a TestSuit than parameterized in MaxNumberOfTestsForEachTestSuite (100))

[image]

sagatowski, 2020-12-20:
Good suggestion!

Beidendorfer, 2021-01-01:
I added a solution in my PR #131

[image]

sagatowski, 2021-01-02:
Solved in 26f40b4.

closed 2021-01-02 by sagatowski