tcunit/TcUnit #129: Unit testing a function block which inherently contains pointers to other function blocks internally causes BSOD

andrewhowick, 2020-12-21:
Hello,

I am trying to unit test a piece of code but every time I run the test, my computer crashes. When I copy and paste the testing procedure outside of the unit test environment and into MAIN, the code runs perfectly fine.

I think the pointers inside of my program could be conflicting with the unit testing library but I can't come to that conclusion. Are you able to look into it from your end and see if that is the case?

Keep up the good work.

sagatowski, 2020-12-21:
@andrewhowick Thanks for your report. Is it possible for you to share a subset of your project that re-creates this issue? If so, could you upload it here?

andrewhowick, 2020-12-21:
Hello Jakob,

I think I know what the problem is, I am running multiple test where each method needs to run multiple cycles. I suspect that a single function block carrying all the different variables is what's causing all my issues.

When I comment out the other test along with its variables and prevent the compiling of its methods respective method, my single run test pass with no problems. Once I uncomment the other test along with its variables, the test would fail again.

[https://1drv.ms/u/s!AolWaJmKVP-GjA4DIQ7N65bJ3NC3?e=fdnRFN](url)

andrewhowick, 2020-12-22:
Nevermind, I separated them into two separate function blocks and the same issue occurred. I can't test both independently nor have them both active at the same time

sagatowski, 2020-12-22:
Is it possible for you to share the failing code? In the link you provided I only see two screenshots

sagatowski, 2020-12-28:
@andrewhowick Have you managed to fix this issue, and if not, is it possible to get a minimal reproduced sample of where this occurs?

andrewhowick, 2021-01-04:
Hello Jakob, sorry for the late reply as I was on holiday.

I have created a similar abit different code to my original source code and have invited you to the repository.

sagatowski, 2021-01-05:
@andrewhowick I've created a separate issue in your repository!

closed 2021-01-22 by sagatowski