tcunit/TcUnit #134: Page Fault when

Arkonos, 2021-01-08:
TcUnit 1.2.0.0 and 1.1.0.0
TwinCAT 4024.10

I have a _FB_massFlowGUI_test_ with 3 methods (sum, noSum, validateInput), each calling other methods of the same FB.
In total a little above 130 tests. I get the following page fault error when I run too many of these tests.
`
'Port_851' (851): Exception (Exception Code: 0xc0000005, Page Fault) in PLC Application PLC Instance, Task PlcTask (RBP: 0xffff9b81ea18e350, RIP: 0xffffc18fe85e67e5, RSP: 0xffff9b81ea18e110)
`
The target then enters an invalid state. I have experienced BSODs when switching back to config mode, be prepared when you test this.

There is no error when I limit the amount of tests. This way I confirmed that no individual test is the cause.
The number of tests that can be uncommented seems not to be static. I had success and trouble with 20, 33, 120 etc.

I have increased the number of tests per FB to 200 and then 500. I've not changed the other parameters.
Most tests are timed. I've changed this time from 0.5 to 10 seconds.
I've tried the new TcUnit.RUN_IN_SEQUENCE without success.
I've kept an eye on memory consumption of the IDE, but no significant increase with plenty of RAM to spare.
I have this error on a second PC with the same software versions.

Here's the archive 
https://www.dropbox.com/s/tpcw0is0r5oh9y3/pageFault.tnzip?dl=0

Any ideas?
Thanks

sagatowski, 2021-01-08:
Hi @Arkonos ! Thanks for your report. Have you tried increasing the GVL_Param_TcUnit.MaxNumberOfTestsForEachTestSuite?

See: https://tcunit.org/frequently-asked-questions/?Display_FAQ=957

Arkonos, 2021-01-08:
Sorry for the title.
Yes, it's currently at 500, 200 didn't help either.

Beidendorfer, 2021-01-08:
@Arkonos  
Did you try to put the ` MaxNumberOfAssertsForEachTestSuite  := 1000;` As the default Value

With the Default values I get this
[image]

@sagatowski We have to Limit the TotalAsserts Array to the max Limits and then create and Error 
Maybe like Issue #132 for the `MaxNumberOfTestsForEachTestSuite`

sagatowski, 2021-01-08:
@Arkonos you do a lot of asserts in every test suite. Adjust `MaxNumberOfAssertsForEachTestSuite` in the parameters and it should be fine.

Arkonos, 2021-01-08:
Of course...
I fooled myself into thinking that I took care of that when I used intermediary methods.
Still wondering what happened when only 33 tests failed, but it seems to work now with the above parameter set to 3000 for good measure.

Thank you for your help!

closed 2021-01-08 by Arkonos

sagatowski, 2021-01-08:
I'm happy it worked out!