tcunit/TcUnit #135: Stack overflow in test can crash system

DrDonut21, 2021-01-12:
It seems that a stack overflow in the code of a unit test can crash the system.
Reproduction steps:
- Create a project with unit tests.
- Implement one test, where the only variable is an array of 0 to 50312 bytes.
- Click 'activate configuration', and check the 'autostart boot project' box
- Your system will now be in a boot loop. (At least, that's what happened to my machine).

If you do not check the 'autostart boot project' box, you can click 'login'. It will now tell you that the Port_851 application does not exist. If you install that, and click run (F5), it will display an error of a stack overflow.

The attached file contains a project with this behavior.
My system:
Windows 10 Enterprise LTSC, 64 bit.
Installed Memory 4,00 GB
TwinCAT version 3.1.4024.7
TcUnit version 1.1.0.0
Please not that this is a virtual machine (using VmWare).

[TcUnitTest.zip](https://github.com/tcunit/TcUnit/files/5801981/TcUnitTest.zip)

If you need any other information, or if there is anything I can do to prevent this behavior, please let me know. I don't know if this is a TcUnit issue, or a TwinCAT issue, so I will try to contact Beckhoff as well...

DrDonut21, 2021-01-12:
Okay, I did a bit more digging around:
- The same problem also occurs on a physical (Non-virtual) machine.
- The problem only occurs if I use an isolated core, not when I use shared cores.
- The problem can be worked around by increasing the maximal stack size in the Realtime settings.

sagatowski, 2021-01-12:
Hi!
Thanks for your report.

The problem is that the stack size in TwinCAT 3 [is fixed to 60KB](https://stackoverflow.com/questions/31292369/is-there-a-way-to-increase-the-stack-size-in-twincat-3). Creating such a huge array in the method fills almost the entire stack.

I simply moved the testArray to the body of the function block instead:
```
FUNCTION_BLOCK FB_CompVSType09_Test EXTENDS TcUnit.FB_TestSuite
VAR
    testArray : ARRAY [0..50312] OF BYTE;
END_VAR
```
Then try to pass a reference to that array to your methods instead, such as:
`M_CheckPointers_NoLvInterlock(testArray);`

I tested this on my machine and it worked fine:

DrDonut21, 2021-01-12:
This huge array was actually because it crashed when we just used a bunch of variables. But I think your suggestion of just putting them in the function block may actually be better.

Thank you for this! I will close the ticket now. The actual crashing of the machine is more of a Beckhoff problem than a TcUnit problem I think...

closed 2021-01-12 by DrDonut21

sagatowski, 2021-01-12:
I'm happy you managed to sort it out!