tcunit/TcUnit #140: AreAllTestsFinished() should be PUBLIC

sagatowski, 2021-02-03:
Hello Jacob,
in  the TcUnit version 1.1.0.0 you declared this method as PUBLIC. " FB_TestSuite.AreAllTestsFinished() "
After the implementation of these functions ( RUN_IN_SEQUENCE(FUN) and TEST_IN_ORDERED(FUN) in a branch, you now declared
this method as INTERNAL. Because i use this method outsite the TcUniit.
Can you declare this method as PUBLIC?
Thank you for the answer.
Have a nice day.
Yok Menns

_Originally posted by @YokMenns in https://github.com/tcunit/TcUnit/discussions/127#discussioncomment-284694_

Roald87, 2021-03-02:
Same for `FB_TestSuite.GetNumberOfTests` and `FB_Test.IsFinished`.

closed 2021-03-06 by sagatowski

Roald87, 2021-03-10:
> Same for FB_TestSuite.GetNumberOfTests, FB_Test.IsFinished and FB_Test.IsFailed. Maybe also more. I think these methods/variables are convenient if the user wants to write some custom methods/function blocks.

What about those?

YokMenns, 2021-03-11:
FB_Test.IsFinished is still INTERNAL.

sagatowski, 2021-03-11:
@YokMenns You wrote:
"in the TcUnit version 1.1.0.0 you declared this method as PUBLIC. " **FB_TestSuite.AreAllTestsFinished()** ""

For FB_Test.IsFinished() you should use `IS_TEST_FINISHED()` instead