tcunit/TcUnit #141: Display of Results in Sequence

kumaraswamygaviyappa, 2021-02-10:
Hello @sagatowski,
I want to Display Test Number and result in Sequence, the code is written as below:

```
      FOR nIndex_ROW:= 1 TO CSV_ROWS BY 1 DO		
		// @TEST-RUN		
		F_IOLinkStateAktuell	:= F_strIOLinkState(IN_State:= nTestInputs);		
		ADSLOGSTR(
					msgCtrlMask := ADSLOG_MSGTYPE_HINT, 
					msgFmtStr   := 'Test Number is %s', 
					strArg      := F_strIOLink_database[nIndex_ROW,0]);
		// @TEST-ASSERT	
		AssertEquals_STRING(
					Expected 	:= sExpectedOutput,
					Actual 		:= F_IOLinkStateAktuell,
					Message 	:= '$' F_IOLinkState $'');	       
     END_FOR	
```
But all **ADSLOGSTR** results displayed first and All  results of **AssertEquals_STRING**.

[image]

Could you please suggest me the way to do it. Thank you.

sagatowski, 2021-02-22:
Hi @kumaraswamygaviyappa! The reason for this is because all things that all ADS-log outputs that TcUnit creates are buffered, the reason for this can be found here:
https://github.com/tcunit/TcUnit/issues/35

TcUnit-Runner (in the current version) unfortunately requires that the test results come in a certain sequence, which the ADS-buffer will guarantee.

I guess this could be solved by adding a way to write to the TcUnit-Ads-logger, but then that would also require changes to TcUnit-Runner.

sagatowski, 2021-03-06:
@kumaraswamygaviyappa Can you look into the required changes to implement this? And if you have time, do a PR.

sagatowski, 2021-03-07:
@kumaraswamygaviyappa

I've taken some time and investigated this further and I think I have a solution to solve your problem. I've added a new function `TCUNIT_ADSLOGSTR()` (takes same arguments as `ADSLOGSTR()`. This function allows to put ADS strings into the TcUnit ADS message buffer. If `Tc2_System.ADSLOGSTR()` is used directly, the messages can come out of sequence in relation to the message created by TcUnit as TcUnit buffers the messages to not overflow the ADS message router. By using this function, the `ADSLOGSTR()` messages are put in the same buffer as TcUnit is using for its output.

I'll do some more tests, add some documentation and do the commit later today.

sagatowski, 2021-03-07:
@kumaraswamygaviyappa I have now fully solved your issue.

Please read [full documentation and how to solve it here](https://tcunit.org/frequently-asked-questions/?Display_FAQ=1011).
This will be included in the 1.2 release of TcUnit. If you want to use it now, please download the complete sourcecode and build the library. Otherwise, I hope to release 1.2 soon.

Added new test-code to the TcUnit-Verifier to verify this functionality. New code is tested with both TcUnit-Verifier and TcUnit-Runner:
```
C:\Code\GitHub_TcUnit\TcUnit\TcUnit-Verifier\TcUnit-Verifier_DotNet\TcUnit-Verifier\bin\Debug>TcUnit-Verifier.exe -v "C:\Code\GitHub_TcUnit\TcUnit\TcUnit-Verifier\TcUnit-Verifier_TwinCAT\TcUnit-Verifier_TwinCAT.sln"
2021-03-07 12:38:51 - Starting TcUnit-Verifier...
2021-03-07 12:38:51 - In Visual Studio solution file, found visual studio version 12.0
2021-03-07 12:38:51 - Loading the Visual Studio Development Tools Environment (DTE)...
2021-03-07 12:39:02 - Cleaning and building TcUnit-Verifier_TwinCAT solution...
2021-03-07 12:39:03 - Generating TcUnit-Verifier_TwinCAT boot project...
2021-03-07 12:39:18 - Activating TcUnit-Verifier_TwinCAT configuration...
2021-03-07 12:39:23 - Restarting TwinCAT...
2021-03-07 12:39:23 - Waiting for TcUnit-Verifier_TwinCAT to finish running tests...
2021-03-07 12:39:33 - ... got 176 report lines so far.
2021-03-07 12:39:43 - ... got 702 report lines so far.
2021-03-07 12:39:54 - ... got 1252 report lines so far.
2021-03-07 12:40:05 - ... got 1382 report lines so far.
2021-03-07 12:40:07 - Asserting results...
2021-03-07 12:40:07 - Done.
2021-03-07 12:40:07 - Closing the Visual Studio Development Tools Environment (DTE), please wait...
2021-03-07 12:40:29 - Exiting application...
```

```
C:\Code\GitHub_TcUnit\TcUnit-Runner\TcUnit-Runner\bin\Release>TcUnit-Runner.exe -v "C:\Code\GitHub_TcUnit\TcUnit\TcUnit-Verifier\TcUnit-Verifier_TwinCAT\TcUnit-Verifier_TwinCAT.sln"
2021-03-07 12:44:13 - TcUnit-Runner build: 0.9.3.0
2021-03-07 12:44:13 - TcUnit-Runner build date: 2021-03-06
2021-03-07 12:44:13 - Visual Studio solution path: C:\Code\GitHub_TcUnit\TcUnit\TcUnit-Verifier\TcUnit-Verifier_TwinCAT\TcUnit-Verifier_TwinCAT.sln
2021-03-07 12:44:13 -
2021-03-07 12:44:13 - In TwinCAT project file, found TwinCAT version 3.1.4022.32
2021-03-07 12:44:13 - In Visual Studio solution file, found visual studio version 12.0
2021-03-07 12:44:13 - Version is pinned: False
2021-03-07 12:44:13 - Trying to load the Visual Studio Development Tools Environment (DTE) version 'VisualStudio.DTE.12.0' ...
2021-03-07 12:44:15 - ...SUCCESSFUL!
2021-03-07 12:44:19 - Using the TwinCAT remote manager to load TwinCAT version '3.1.4024.12'...
2021-03-07 12:44:25 - No task name provided. Assuming only one task exists
2021-03-07 12:44:25 - Found task with name 'PlcTask'
2021-03-07 12:44:33 - Setting target NetId to '127.0.0.1.1.1'
2021-03-07 12:44:33 - Enabling boot project and setting BootProjectAutostart on 127.0.0.1.1.1
2021-03-07 12:44:57 - Waiting for results from TcUnit...
2021-03-07 12:45:07 - ... got 173 report lines so far.
2021-03-07 12:45:17 - ... got 674 report lines so far.
2021-03-07 12:45:28 - ... got 1224 report lines so far.
2021-03-07 12:45:39 - ... got 1383 report lines so far.
2021-03-07 12:45:40 - All results from TcUnit obtained
2021-03-07 12:45:50 - Done collecting TC results
2021-03-07 12:45:50 - Writing xUnit XML file to C:\Code\GitHub_TcUnit\TcUnit\TcUnit-Verifier\TcUnit-Verifier_TwinCAT\TcUnit_xUnit_results.xml
2021-03-07 12:45:50 - Closing the Visual Studio Development Tools Environment (DTE)...
2021-03-07 12:46:10 - Exiting application...
```

closed 2021-03-07 by sagatowski

kumaraswamygaviyappa, 2021-03-08:
@sagatowski Thank you so much. It will be of great help.

I could also get what i wanted using pointers.
```
pt := ADR(sTestMessage);
AssertEquals_BOOL(
		Expected  := bStartenErwartet,
                Actual      := bStartenAktuell,
                Message   := pt^);
```