tcunit/TcUnit #145: Improve documentation/API of LogExtendedResults

Roald87, 2021-03-02:
I was playing around with v1.2 and it took a very long time before the test results were printed. This was caused by all the messages which were printed. Only after some digging I found the global variable `LogExtendedResults` with which I could disable this.

I think it is currently very unclear to the user where these messages come from and how to disable them. Maybe the default of `LogExtendedResults` should be false and only the users who want to use the TcUnit-Runner need to enable it? Or at least add something to the FAQ/documentation about this parameter. Or is this already there? I couldn't find any at least.

sagatowski, 2021-03-02:
Hi! I agree with you. When I added the new data for the integration of TcUnit-Runner (from version 1.1), I noticed that with very large projects it added quite some time for all the outputs to be printed. The reason for this is that I needed to put everything in a buffer, then dispatching them to ADS one by one with a time-delay before, as some ADS-messages would be lost otherwise (there is one or two closed issues in the system that document this). Now, the rationale for having this enabled by default was:
1. I assumed that most libraries that have tests won't have too many tests (whatever too many is, but 500+ ~ish), so the performance hit wouldn't be too big
2. If it was disabled by default, I was afraid that TcUnit-Runner wouldn't work for a lot of people, and that it would result in me (or GitHub issues) getting spammed with questions. 

Now of course that TcUnit-Runner is released (at least in a early release), and TcUnit 1.1 is released, I'm even more afraid to go back as it would change the default behavior 

I agree that whichever route should be default (which to be honest, my gut-feeling doesn't even guide me right now), it should be included in the F.A.Q., which it is currently not. I will add an entry to the F.A.Q. this weekend.

Roald87, 2021-03-02:
> I assumed that most libraries that have tests won't have too many tests (whatever too many is, but 500+ ~ish), so the performance hit wouldn't be too big

Mine was not that big (100 ish), but the cycle time was set to 50 ms, so I guess that is why it took a while.

>  in me (or GitHub issues) getting spammed with questions.

😁 

> I will add an entry to the F.A.Q. this weekend.

👍

sagatowski, 2021-03-06:
I have added an entry to the FAQ:
https://tcunit.org/frequently-asked-questions/?Display_FAQ=1002

I hope this is clear enough.

closed 2021-03-06 by sagatowski