tcunit/TcUnit #149: FB_AdsLogStringMessageFifoQueue.TcPOU, FB_MemRingBuffer, array struct

runtimevic, 2021-04-28:
Hello Jakob,

[image]

in the declaration --> ArrayBuffer : ARRAY [0..RING_BUFFER_SIZE] OF ST_ADSLogStringMessage;
I think this statement would be better but I can't get it to work? for example I'm trying to do a test project first

https://infosys.beckhoff.com/english.php?content=../content/1033/tcplclibutilities/html/TcPlcLibUtilities_FB_MemRingBuffer.htm&id=

replacing the byte array with a structure array.

thank you, Víctor.

sagatowski, 2021-04-28:
Hi Victor!

I don't understand... what would be better?

The Tc2_Utilities.MemRingBuffer can only work with a BYTE-array, that's why the buffer is declared as a BYTE array.

runtimevic, 2021-04-29:
> ¡Hola Victor!
> 
> No entiendo... ¿Qué sería mejor?
> 
> El Tc2_Utilities.MemRingBuffer solo puede funcionar con una matriz BYTE, es por eso que el búfer se declara como una matriz BYTE.

Hello Jakob,
That is the problem that the array has to be declared as a BYTE data type and cannot be an ST data structure, 
will you not know how it could be done for the array to be of the data structure type?
thanks

sagatowski, 2021-04-30:
But why does the array have to be declared as ST data structure? What problem are you trying to solve?

As I said, the FB_MemRingBuffer only accepts a BYTE-array as storage:
https://infosys.beckhoff.com/english.php?content=../content/1033/tcplclibutilities/html/TcPlcLibUtilities_FB_MemRingBuffer.htm&id=

This is a Beckhoff function block, not a TcUnit function block.

"pBuffer: Address of PLC variable (e.g. ARRAY[...] OF BYTES ) that is to be used by the function block as buffer storage. The address can be determined with the ADR operator."

runtimevic, 2021-04-30:
> But why does the array have to be declared as ST data structure? What problem are you trying to solve?
>
> As I said, the FB_MemRingBuffer only accepts a BYTE-array as storage: [...]

Hello Jakob.
if everything is very clear,
It was to see if I could have the buffer array as a ST structure, so it could be observed and be visualized.
thanks

sagatowski, 2021-05-02:
Aha now I understand! I think the easiest way in that case is to store everything in a local structure array, and copy it over to the byte-buffer. In other words, have them side-by-side!

runtimevic, 2021-05-03:
> Aha now I understand! I think the easiest way in that case is to store everything in a local structure array, and copy it over to the byte-buffer. In other words, have them side-by-side!

Hi Jakob,
and you couldn't replace FB_MemRingBuffer with MEMCPY?
thanks you.

sagatowski, 2021-05-04:
I can't see how FB_MemRingBuffer can be simply replaced with a MEMCPY?
I think a new FB_MemRingBuffer would have to be developed that can store any data as a buffer and not just a byte-buffer. As you say, a MEMCPY would probably be used but there would be something inbetween to make it a ring buffer.

Why do you need to visualize this buffer?

runtimevic, 2021-05-04:
> I can't see how FB_MemRingBuffer can be simply replaced with a MEMCPY?
> I think a new FB_MemRingBuffer would have to be developed that can store any data as a buffer and not just a byte-buffer. As [...]

Hello Jakob,
Because I saw your code and wanted to apply it to make a fifo with a data structure and it seemed like a good idea to use it FB_MemRingBuffer, won't you have a developed FB_fifo?

sagatowski, 2021-05-04:
So you wanted to re-use the code in another application? Do I understand correctly?
In that case I would recommend to write your own FIFO that uses your custom structure!

runtimevic, 2021-05-05:
> So you wanted to re-use the code in another application? Do I understand correctly?
> In that case I would recommend to write your own FIFO that uses your custom structure!

Hello Jakob,
if I would like to have an array formed by a data structure and replace the FB_MemRingBuffer and with this get a FB_FIFO, but I do not know that fb lean to perform it since those I have found rely on BYTES.

sagatowski, 2021-05-06:
Hi @runtimevic !

This probably isn't the right place to ask for a custom implementation of FB_FIFO. I can highly recommend you StackOverflow (mark the question with TwinCAT) or the TwinCAT linkedin group, maybe someone there might have some ideas of implementing a custom FIFO tailored for a structure! Here you have some links: http://alltwincat.com/links/

runtimevic, 2021-05-07:
> ersonalizado adaptado para una estr

> Hi @runtimevic !
>
> This probably isn't the right place to ask for a custom implementation of FB_FIFO. I can highly recommend you StackOverflow (mark the question with TwinCAT) or the TwinCAT linkedin [...]

Hi @sagatowski , thanks you.

closed 2021-05-08 by sagatowski