tcunit/TcUnit #152: Idea: include .library in release

Roald87, 2021-04-30:
I'm currently upgrading a project from 1.0 to 1.2. Version 1.2 was installed with the .compiled-library version. Now I'm experiencing some issues and want to figure out how to solve them. For this it would be useful to have the .library version, such that I can step into the code when debugging. 

Furthermore it would be useful in general. For example when I'm curious how a function is implemented. Now if I want to see the source code I have to clone the project, or look on GitHub.

I'm not sure if there are any drawbacks/differences between the .compiled-library and .library version, apart from hiding the source code. [Infosys only mentions](https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_plc_intro/4189255051.html&id=):

> To make library modules view- and access-protected, a library project can be saved in a precompiled format (<project name>.compiled-library) – see Command Save as library.

Since TcUnit is open source, I do not see the need to hide the source code. But maybe I'm missing something.

Of course I can compile my own version, but for convenience it would be nice to include this in the release 😄

Roald87, 2021-05-03:
I send support an email.

Roald87, 2021-05-04:
I got an answer! The .library files are bigger because they contain all source code. Apart from that there are no differences.

Roald87, 2021-05-04:
Found a bit more info on [this InfoSys page](https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_plc_intro/4189333259.html&id=)

> `*.library` (source library)
>  - You can open a source library (for viewing and/or editing) by using the command Add Existing Item, which is available on the PLC node within the project tree.
>  - You can "step" into a source library using the usual debug functionality.
> 
> `*.compiled-library` (compiled library)
> - This file extension can be used to save a library project in a compiled format. An encrypted image of the precompile context of the library is stored, which means the implementations of the library function blocks are no longer accessible or visible.
> - A compiled library cannot be opened or debugged.
> - Otherwise, *.compiled-library files behave like *.library files. You can therefore install and reference them in the same way.
> - The source code of a library can be protected by using the compiled library format. In addition, the library files are smaller and the loading times are shorter.
>

sagatowski, 2021-05-04:
https://tcunit.org/introduction-user-guide/

Needs to be updated

sagatowski, 2021-05-04:
You've convinced me 👍 

Latest release has been updated: https://github.com/tcunit/TcUnit/releases/tag/1.2.0.0

Documentation updated as well: https://tcunit.org/introduction-user-guide/

Thanks!

closed 2021-05-04 by sagatowski