tcunit/TcUnit #154: Visual studio path

kumaraswamygaviyappa, 2021-06-14:
Hallo,

The VS path was found as marked in RED line but latter Error as Marked in Orange.
Any clues please.

[image]

sagatowski, 2021-06-14:
`LaunchPlcStaticAnalysis.bat?` This does not seem to be related to TcUnit?

kumaraswamygaviyappa, 2021-06-14:
Hi @sagatowski ,

Yes, But any help is much appreciated as i am struggling from this from few days.
`LaunchPlcStaticAnalysis.bat` works fine when executed from CMD.
Error occurs only while using Jenkins.

sagatowski, 2021-06-14:
Then my guess is that this is a privilegesproblem. Have you made sure that the Jenkins agent has the same privileges as the user running from the CMD?

kumaraswamygaviyappa, 2021-06-14:
I couldn't find the privileges , i could solve it by changing `JENKINS_HOME`. 
Could give me some pointers  to find the privileges in Jenkins.

sagatowski, 2021-06-14:
Jenkins_HOME? How did that change the problem? So what was the problem then?

I guess we can close this issue then?

kumaraswamygaviyappa, 2021-06-14:
Followed This Link [JENKINS_HOME](https://stackoverflow.com/questions/34854377/how-to-change-workspace-and-build-record-root-directory-on-jenkins)

closed 2021-06-14 by kumaraswamygaviyappa

sagatowski, 2021-06-14:
Thanks @kumaraswamygaviyappa !

I have also written a little about this:
https://alltwincat.com/2018/07/26/ci-cd-with-twincat-part-two/

And more importantly:
https://alltwincat.com/2018/10/04/ci-cd-with-twincat-part-four/

(search for the word "weekend" in the second link and you will find what I mean)