tcunit/TcUnit #155: Unhandled Exception

kumaraswamygaviyappa, 2021-06-14:
Hallo,

The project is working in my system. When the same project is executed in the server I am facing this issues.
I am getting an Unhandled Exception in line 182.
 

[image]

sagatowski, 2021-06-14:
Hi @kumaraswamygaviyappa!

This is a little confusing.
1. First of all this doesn't seem to have anything to do with TcUnit nor TcUnit-Runner? TcUnit is just a TwinCAT library project, not a C# application.
2. This is TcStaticCodeAnalysisLoader... which is this project: https://github.com/sagatowski/TcStaticAnalysisLoader ?
3. Even looking at the above repo, and looking at the correct file, this is the wrong line (look at line 182): https://github.com/sagatowski/TcStaticAnalysisLoader/blob/master/TcStaticAnalysisLoader/TcStaticAnalysisLoader/Program.cs

In other words, it's just plain impossible for me to know exactly how to reproduce this error, and please put it in the right place.
I kindly ask you to read this guide: https://medium.com/nyc-planning-digital/writing-a-proper-github-issue-97427d62a20f

closed 2021-06-14 by sagatowski

kumaraswamygaviyappa, 2021-06-16:
Hello @sagatowski ,

You're right, its nothing do with TcUnit but i posted here hoping to find solution here.
Please if possible please to look in to this.

the same code and project works well in setup2

[image]

sagatowski, 2021-06-16:
Shooting in the dark here as I don't have enough information, but I would guess it has to do with permissions:
https://alltwincat.com/2018/10/04/ci-cd-with-twincat-part-four/

(search for the word "weekend" in the second link and you will find what I mean)

kumaraswamygaviyappa, 2021-06-16:
Sorry for not providing enough information. Could you please let me know the further information required for you to understand this issue.

- I am trying to execute [TcStaticCodeAnalysisLoader](https://github.com/sagatowski/TcStaticAnalysisLoader)
- In my Laptop with `win7`, `TcUnit` `TcRunner` and `Jenkines` all works as expected
- same project which is running in my PC was executed on another machine running on  `winserver 2019` with all same versions of software.
- `LaunchPLCStaticAnalysis.bat` was executed in `CMD` and got error as mention below.

`Unhandled Exception: System.ArgumentNullException: Value cannot be null.
Parameter name: type
   at System.Activator.CreateInstance(Type type, Boolean nonPublic)
   at AllTwinCAT.TcStaticAnalysisLoader.Program.Main(String[] args) in D:\TcStaticAnalysisLoader\TcStaticAnalysisLoader\Program.cs:line 182`

- As you mentioned, i am looking in to permissions in `winserver 2019`

sagatowski, 2021-06-17:
With what parameters did you execute the batch-script?

kumaraswamygaviyappa, 2021-06-17:
Hello @sagatowski,
Thank you for help me.

Parameters:
`"%TC_STATIC_ANALYSIS_LOADER_PATH%" -v %VISUAL_STUDIO_SOLUTION_PATH% -t %TWINCAT_PROJECT_PATH% -b %BUILD_OR_TE1200%`

`BUILD_OR_TE1200` is new parameter for our requirement.

C# code is also updated:
```
        .Add("v=|VisualStudioSolutionFilePath=", v => VisualStudioSolutionFilePath = v)
        .Add("t=|TwinCATProjectFilePath=", t => TwinCATProjectFilePath = t)
        .Add("b=|BuildOrTE1200OrUtest=", b => BuildOrTE1200OrUtest = b)
        .Add("?|h|help", h => showHelp = h != null);
```
I tried to debug C# code further in VS2019 in both machines by providing absolute paths.
```
        private static string VisualStudioSolutionFilePath = "C:\\Test\\Test.sln";
        private static string TwinCATProjectFilePath = "C:\\Test\\Test\\Test.tsproj";
        private static string BuildOrTE1200OrUtest = "BUILD";
```
Observed different `type `for `Class System Type`:
**Machine (win2007)(works fine):**
        `Class System Type` observed as
         `Value: {Name = "__ComObject" FullName = "System.__ComObject"}`	
        `Type: System.Type {System.RuntimeType}`
	

[image]

**Machine (winServer 2019)(Facing issue):**
        `Class System Type` observed as
         `Value: null`	
        `Type: System.Type`

[image]

sagatowski, 2021-06-17:
Hi @kumaraswamygaviyappa!
Do you have VS2017 installed on the Windows 2019 server machine?

kumaraswamygaviyappa, 2021-06-17:
Yes,
On both machines,

```
Microsoft Visual Studio Community 2017 
Version 15.9.36
VisualStudio.15.Release/15.9.36+28307.1525
Microsoft .NET Framework
Version 4.8.03761

Installed Version: Community
```

sagatowski, 2021-06-17:
What was your conclusion about permissions in Windows server 2019? I can't find it in this discussion.

kumaraswamygaviyappa, 2021-06-17:
Hello @sagatowski ,

I didn't do much on the permissions as i felt its with VS( am i right?). But I have admin rights.

Do i need any other packages to be installed?

[image]

kumaraswamygaviyappa, 2021-06-17:
Hello @sagatowski ,

Sorry for troubling you.
I have installed many more packages after looking in to VS installation into the machine(win2007) which is working.
Now machine with winServer 2019 is also working. But i don´t now which package solved the problem. 
```
Microsoft Visual Studio Community 2017 
Version 15.9.36
VisualStudio.15.Release/15.9.36+28307.1525
Microsoft .NET Framework
Version 4.8.03761

Installed Version: Community

ASP.NET and Web Tools 2017   15.9.04012.0
ASP.NET and Web Tools 2017

ASP.NET Core Razor Language Services   15.8.31590
Provides languages services for ASP.NET Core Razor.

ASP.NET Web Frameworks and Tools 2017   5.2.61435.0
For additional information, visit https://www.asp.net/

Azure App Service Tools v3.0.0   15.9.03024.0
Azure App Service Tools v3.0.0

C# Tools   2.10.0-beta2-63501-03+b9fb1610c87cccc8ceb74a770dba261a58e39c4a
C# components used in the IDE. Depending on your project type and settings, a different version of the compiler may be used.

Common Azure Tools   1.10
Provides common services for use by Azure Mobile Services and Microsoft Azure Tools.

JavaScript Language Service   2.0
JavaScript Language Service

Microsoft Continuous Delivery Tools for Visual Studio   0.4
Simplifying the configuration of Azure DevOps pipelines from within the Visual Studio IDE.

Microsoft JVM Debugger   1.0
Provides support for connecting the Visual Studio debugger to JDWP compatible Java Virtual Machines

Microsoft Library Manager   1.0
Install client-side libraries easily to any web project

Microsoft MI-Based Debugger   1.0
Provides support for connecting Visual Studio to MI compatible debuggers

Microsoft Visual Studio Tools for Containers   1.1
Develop, run, validate your ASP.NET Core applications in the target environment. F5 your application directly into a container with debugging, or CTRL + F5 to edit & refresh your app without having to rebuild the container.

NuGet Package Manager   4.6.0
NuGet Package Manager in Visual Studio. For more information about NuGet, visit http://docs.nuget.org/.

ProjectServicesPackage Extension   1.0
ProjectServicesPackage Visual Studio Extension Detailed Info

ResourcePackage Extension   1.0
ResourcePackage Visual Studio Extension Detailed Info

ResourcePackage Extension   1.0
ResourcePackage Visual Studio Extension Detailed Info

SQL Server Data Tools   15.1.62002.01090
Microsoft SQL Server Data Tools

TcProjectCompare   1.0
Visual Studio Integrated Version of TcProjectCompare

TcXaeDebuggerLiveWatch   1.0
TwinCAT XAE Live Watch Tool Window

TcXaeHelper   1.0
Info

TcXaeModules   1.0
TwinCAT TMC Editor, TwinCAT TMC Code Generator

TcXaeStartPage   1.12
Beckhoff Start Page for TwinCAT XAE

TwinCAT XAE Base   3.1.0.0
The TwinCAT XAE Base extension provides the basic project structure for TwinCAT projects.

TwinCAT XAE EventLogger   1.0
The TwinCAT XAE Eventlogger extension provides the capability of monitoring events sent through the TwinCAT 3 Eventlogger of a target system.

TwinCAT XAE PLC   3.1.0.0
PLC Environment

TypeScript Tools   15.9.30718.2001
TypeScript Tools for Microsoft Visual Studio

Visual Basic Tools   2.10.0-beta2-63501-03+b9fb1610c87cccc8ceb74a770dba261a58e39c4a
Visual Basic components used in the IDE. Depending on your project type and settings, a different version of the compiler may be used.

Visual Studio Code Debug Adapter Host Package   1.0
Interop layer for hosting Visual Studio Code debug adapters in Visual Studio

Visual Studio Tools for Containers   1.0
Visual Studio Tools for Containers
```

sagatowski, 2021-06-17:
Hi @kumaraswamygaviyappa ! Thanks for feedback. I'm very happy you got it to work!
Maybe one of the packages is the one that gives the DTE functionality?

kumaraswamygaviyappa, 2021-06-17:
Hallo @sagatowski ,
I think so, but i am sorry, I don't have any clue. 

The trouble from me will continues as DTE problem got solved but "remote manager" thing giving error. But from batch file it works fine.

I remember "Weekend" keyword from CI/CD Article and go through in the morning and update here.

Thank you very much for your patience.

sagatowski, 2021-06-17:
Remember that the remote manager only works from 4022 if I remember correctly.
Crossing my fingers you will sort it out.

kumaraswamygaviyappa, 2021-06-21:
after Job is executed in one of the Jenkins client, everything is working well.
Thank you @sagatowski for all your inputs.

sagatowski, 2021-06-21:
@kumaraswamygaviyappa I'm happy to hear you figured it out. I didn't have a doubt for a single second that you wouldn't 😄