tcunit/TcUnit #156: Multi-cycle tests

aldoa-cellares, 2021-07-15:
https://github.com/tcunit/TcUnit/blob/df90eadd6444045b75620663b71d30f2ba2e0027/TcUnit/TcUnit/POUs/FB_TcUnitRunner.TcPOU#L59

I was just wondering if I am understanding the intent of the TcUnitRunner correctly. If a test suite takes multiple cycles to execute it seems that the NumberOfTestSuitesFinished counter will continue to increment for any test suites that have already completed while waiting for the multi-cycle tests to complete. The flag AllTestSuitesFinished only gets set if the number of finished tests is exactly equal to the number of test suites. This seems problematic with the above code as it will continue to increment and go past the total number of possible test suites to finish. In fact the counter can ever overflow and wrap around. Would it be possible to add code to just record if the test finished for a suite and only increment for that test once?

There seems to be an issue depending on the order that the test suite instances are instantiated. Specifically it seems that single cycle tests cannot be called after the last multi-cycle test? Depending on if my multi-cycle test is the last declared variable or not breaks the unit tester. At least this is the issue that I am running into.

sagatowski, 2021-07-22:
Can you upload an example that re-creates the issue, and post it here?

sagatowski, 2022-04-18:
Closed due to inactivity. You are welcome to re-open the issue.

closed 2022-04-18 by sagatowski