tcunit/TcUnit #157: Total number of asserts

Alec-Mack, 2021-07-20:
Hi  @sagatowski 

I'm really impressed with the TcUnit suite of tools.  Thank you for creating them.

Would it be possible to return the total number of asserts in the test summary please? (The Error ADS Log table you create) Or is there an option for this that I haven't found?

Many thanks

Alec

sagatowski, 2021-07-22:
Hi @Alec-Mack!
Is there any particular reason why you would want to display this data?

The data is available as a variable. You can check the variable `AssertResultInstances` in every instance of `AssertResults` of every test suite (`FB_TestSuite`). Check the description of `FB_AssertArrayResultStatic`. I think this should give you the information you need.

Alec-Mack, 2021-07-22:
Classification:UNCLASSIFIED

Hi Jakob

I would like to display it as it is a big number and I think it would give me a warm feeling seeing it grow as I write more tests.

Alec

Alec Mackendrick BEng, MIET CEng | Principal Control Systems Engineer
Defence Systems Technology | Marine & Technology
Babcock International Group
Ashton House | Ashton Vale Road | Bristol | BS3 2HQ
Tel: +441174 408 514 | ***@***.******@***.***>
www.babcockinternational.com<http://www.babcockinternational.com>

P Please consider the environment before printing this email

This electronic mail message, including any attachments, is a confidential communication exclusively between Babcock International and the intended recipient(s) indicated as the addressee(s). It contains information which is private and may be proprietary or covered by legal professional privilege. If you receive this message in any form and you are not the intended recipient you must not review, use, disclose or disseminate it. We would be grateful if you could contact the sender upon receipt and in any event you should destroy this message without delay. Anything contained in this message that is not connected with the business of Babcock International is neither endorsed by nor is the liability of Babcock International.

Babcock International Group PLC.

Website: www.babcockinternational.com
Registered in: United Kingdom
Registration No: 6717269
33 Wigmore Street, London, W1U 1QX

sagatowski, 2021-07-22:
Hi Alec.

For this reason it's enough that you see the amount of succesful and failed tests.
You should anyway strive to have only one (or few) assert per test. Please see:

https://softwareengineering.stackexchange.com/questions/7823/is-it-ok-to-have-multiple-asserts-in-a-single-unit-test

Alec-Mack, 2021-07-22:
Classification:UNCLASSIFIED

Thanks Jakob

I’m new to Unit testing so I’m not familiar with some of the philosophies.  Every day is a learning day.

Alec

https://softwareengineering.stackexchange.com/questions/7823/is-it-ok-to-have-multiple-asserts-in-a-single-unit-test

sagatowski, 2021-07-22:
Hey Alec!

No problem, everyone is new to everything at some point :)

If you think this answer is OK, can you close the issue?
Thanks.

/ Jakob

Alec-Mack, 2021-09-03:
Hello Jakob
I hope you are well.
I've run into the problem that Arkonos reported here:  https://github.com/tcunit/TcUnit/issues/134  where many asserts caused a PageFile issue and BSODs.
I've increased the 3 TcUnit MaxNumberOf... parameters, and while the tests now pass, I still get a BSOD.
I have many asserts which make very repetitive checks and I am going through my tests to make sure that I don't do asserts on these by doing this sort of thing:
```
IF NOT (Result = LastResult AND Expected = LastExpected) THEN
		AssertEquals(Expected := Expected,
				 Actual := Result,
				 Message := 'Display fault message.');
END_IF
```

Which seems to be doing the trick and reducing the number of asserts.  Now I would like to know the total assert count, not for a warm fuzzy feeling, but to make sure that I don't approach conditions that might BSOD my development machine.

From reading your documentation for `FB_AssertResultStatic` it looks like I shouldn't need to do the above.  Maybe something isn't quite right in there or maybe I've misunderstood?  I know Unit testing philosophy says the number of asserts should be low, but I have conditions in the machine the PLC will eventually control that need to persist for reasonable (in human terms) amounts of time.

To collect the number of asserts, I tried following your advice, but `AssertResults` doesn't seem to be a method of my FB_TestSuite.

`FB_Test.GetNumberOfAssertions()` looks like it could help, but it is INTERNAL and is keeping its information to itself :-).

Are you able to provide any further advice please?

Many thanks

Alec

sagatowski, 2022-04-18:
Hi! Sorry for very late reply. What type of PLC are you running the tests on? How much AMS/Router memory is allocated to the runtime?
Might it simply be a memory problem? That's the only reason I can think of that you get BSOD.

sagatowski, 2022-04-18:
@Alec-Mack, have you checked this issue? https://github.com/tcunit/TcUnit/issues/148

Alec-Mack, 2022-04-21:
Hi @sagatowski 
It might have been a memory problem.  I left the 'If nothing has changed, don't bother doing the assert' check and everything behaved itself.  I've moved onto other things, so I'll close the issue.
Thanks for the support.
Alec

closed 2022-04-21 by Alec-Mack