tcunit/TcUnit #158: TcUnit testing a Method within a FB.

mariushmelaas, 2021-07-23:
Hi! In our project we use many FB, and even more Methods within the FB. Is it possible to test these? I've tried with a really easy logic to test myself, but with no luck.

sagatowski, 2021-07-23:
Hi @mariushmelaas!

It's done in the same way as the shown in the examples for function blocks.

So you basically define your test as usual:

```
VAR
    fbMyFunctionBlockWithMethods : FB_MyFunctionBlocksWithMethods;
    nResult : INT;
END_VAR

TEST('MyTest');

// Call method
nResult := fbMyFunctionBlockWithMethods.Sum(a := 1, b := 2);

AssertEquals_INT(Expected := 3,
                            Actual := nResult,
                            Message := 'Result not correct!');

TEST_FINISHED();
```

mariushmelaas, 2021-07-23:
Thanks, now it works! :)

closed 2021-07-23 by mariushmelaas