tcunit/TcUnit #160: Missing test results

bernienor, 2021-09-14:
The summary of test results disappeared from my project. Failing tests would still show up. After some investigation I think I found the cause.

If the TEST macro called with a text string ending in a space, the test result summary for the project will fail to print.

This fails:
TEST('Home command parsing test ');

This works as it should:
TEST('Home command parsing test');

Using TcUnit 1.2.0.0

bernienor, 2021-09-14:
Another observation:

When it fails, the tests are run continuously. I guess the missing summery is a result of the tests never finish.

sagatowski, 2021-09-30:
Thanks for this report. I can confirm that I get the same result.
This will have to be investigated.

sagatowski, 2021-09-30:
@bernienor I'm onto a solution. I'll let you know when it's done.

sagatowski, 2021-09-30:
Issue solved (see commit 5059342). @bernienor  You can create TcUnit from source if you want this feature, I'll wait with a release until I've fixed some other issues.

closed 2021-09-30 by sagatowski

bernienor, 2021-09-30:
Thanks!

As long as I know how to fix this in my code I'm happy. 

Cheers
Bernie

sagatowski, 2021-09-30:
Great, appreciate that you found this bug! Thanks.