tcunit/TcUnit #161: Tests that should fail are not

MrSamuelLaw, 2021-09-20:
Hi, when running tests sequentially, tests that should fail are not, for example, the output below includes the test shown underneath it. It is included in the discovered tests, and it as you can see, TcUnit has somehow determined that it passed.

Severity	Code	Description	Project	File	Line	Suppression State
Error		9/20/2021 4:07:54 PM 516 ms | 'PlcTask' (350): | ==========TESTS FINISHED RUNNING==========				
Error		9/20/2021 4:07:54 PM 536 ms | 'PlcTask' (350): | Test suites: 2				
Error		9/20/2021 4:07:54 PM 556 ms | 'PlcTask' (350): | Tests: 8				
Error		9/20/2021 4:07:54 PM 576 ms | 'PlcTask' (350): | Successful tests: 8				
Error		9/20/2021 4:07:54 PM 596 ms | 'PlcTask' (350): | Failed tests: 0				
Error		9/20/2021 4:07:54 PM 616 ms | 'PlcTask' (350): | ======================================	

TEST_ORDERED('failing test');
	THIS^.AssertTrue(FALSE, 'mesage');
TEST_FINISHED();	

In my humble opinion, this is catastrophic for a unit testing library to allow failing tests to pass. I have used many frameworks, gtest, python's built-in & pytest, and most recently mocha + chia and have never had something like this happen... so I am requesting two things. 

1. The bug be fixed as this turns TcUnit into a bug factory.
2. Point me in the right direction to maybe help, or find the root cause of the bug on my own, so I can create a pull request and not just complain, but actively work to fix the issue for myself and others going forward.	

Thank you in advance

Dev enviroment
OS: Windows 10 XAE
TwinCat v3.1.4024.17
TcXaeShell version 15.0.28010.2050 D 15.8

Runtime enviroment
Beckhoff PLC CX 5130
TwinCat v3.1.4024.11

HAHermsen, 2021-09-20:
This is NO bug, IMHO you are calling the ordered test wrong:

```
IF TEST_ORDERED('failing test') THEN
   THIS^.AssertTrue(FALSE, 'mesage');
   TEST_FINISHED();
END_IF;
```

Should do the trick ;-)

MrSamuelLaw, 2021-09-20:
IMHO, You're totally right, the example I found previously was missing the IF statements.

sagatowski, 2021-09-21:
@MrSamuelLaw Where is the wrong example? I can't find it in the documentation.

Can this ticket be closed?

PTKu, 2021-09-21:
Yeah! This ticket should not be opened with such a strong prelude in the first place. IMHO.

closed 2021-09-21 by MrSamuelLaw