tcunit/TcUnit #177: Library SysDir, 3.5.8.0 (System) not installed, or no license found

marcusl, 2022-01-17:
I get this both when using the compiled code with a test and when building TcUnit from source.

Google didn't give me much to go on trying to google for this, neither is this lib in the _Libraires folder in the downloaded source, but some other system libs are.

I'm using TwinCAT XAE Shell and TwinCAT Build 4024.22. 

All help greatly appreciated! This seems to be a great project!

sagatowski, 2022-01-17:
Hi Marcus!

I can't quite figure out the problem description.
Have you activated trial licenses?

Do you have a screenshot?

What version of TcUnit are you running?

TcUnit 1.2 on 3.1.4024.22 works fine here.

marcusl, 2022-01-18:
Hi Jakob,

TcUnit: I used the 1.2 compiled library, but with code it was the latest master from Jan 7th c9c650f4a0adc0847d011d5df6984b790f12947c  (freshly cloned submodule) 

```
C:\Dev\CatFish\src\3rdparty\TcUnit [(c9c650f...) +0 ~1 -0 !]> git log -1
commit (HEAD) c9c650f4a0adc0847d011d5df6984b790f12947c 
Author: Jakob Sagatowski <jakob.sagatowski@outlook.com>
Date:   Fri Jan 7 19:20:09 2022 +0000

    Fixed some formatting.
    Upped log4net to 2.0.14 for the verifier.
```

[image]

Hm. License looks good, TC3 PLC, TC3 NC PTP , TC3 NCI.  Now my PLC project with test compiles, but not the TcUnit source.

FahrJo, 2022-01-18:
As far as I know this is a CODESYS lib which is not supported by Beckhoff and they also advise against using it and recommend their FBs instead (https://infosys.beckhoff.com/content/1031/tcplclibsystem/html/tcplclibsys_fileio_example.htm?id=4724517592797591316).
I believe that it was shipped with early TwinCAT versions, but it may be that it is no longer included in newer TwinCAT versions.

sagatowski, 2022-01-18:
Hi @marcusl !
I'm honestly not sure exactly why you've tried to add the SysDir manually?

I did this:
- Created a fresh virtual machine
- Installed TwinCAT 3.1.4024.22 on it
- Downloaded [TcUnit 1.2](https://github.com/tcunit/TcUnit/releases/tag/1.2.0.0)
- Installed the library
- Created a new TwinCAT 3 project
- Referenced the above library
- It all compiled & works
[image]

Did you follow this guide?
https://tcunit.org/introduction-user-guide/

(Install from library file)

sagatowski, 2022-01-18:
> As far as I know this is a CODESYS lib which is not supported by Beckhoff and they also advise against using it and recommend their FBs instead [...]

The reason that it's using SysDir etc is in order for it to be compatible with [counit] (https://forge.codesys.com/lib/counit/home/Home/).

(it's a fork of TcUnit)

marcusl, 2022-01-18:
```
Did you follow this guide?
https://tcunit.org/introduction-user-guide/

(Install from library file)
```

To the best of my ability. :)   I will try again.

sagatowski, 2022-01-18:
> ```
> Did you follow this guide?
> https://tcunit.org/introduction-user-guide/
> 
> (Install from library file)
> ```
> 
> To the best of my ability. :) I will try again.

Let me know if anything is unclear and I will try to make it more clear.

marcusl, 2022-01-18:
I probably got myself into a pickle first trying to add the TcUnit project from source, seeing the sysdir error, trying to fix it, and then it went downhill from there.

sagatowski, 2022-01-18:
@marcusl 
Does it work now? In that case close the issue.
You are welcome to [star the project](https://github.com/tcunit/tcunit/stargazers) if you like it!

marcusl, 2022-01-18:
Starred! :)  Give me a few moments, I will update/close issue one I have poked it a bit.

FahrJo, 2022-01-18:
> The reason that it's using SysDir etc is in order for it to be compatible with [counit](https://forge.codesys.com/lib/counit/home/Home/).
>
> (it's a fork of TcUnit)

I already wondered and wanted to suggest that I can update that at some time... Thanks for the clarification!

HAHermsen, 2022-01-18:
Hi,

There are at two senario's possible: or you let TcUnit PLC code create a report internally, or you let the TcUnit-Runner (C#) create it. TcUnit defaults to the TcUnit-Runner. Both options are open for TcUnit users which provides a scalable solution.

Regards

marcusl, 2022-01-18:
I have SysDir 3.5.12.0 but the error is about 3.5.8.0, which is what the tc library refers to. If works if I update    But I see for your VM it auto-updated the ref to 3.5.12.0.   

However, if I update the resolution to 3.5.12.0 it seems to work:

[image]

I also had to change the build target to "TwinCAT RT", this might've been the main issue.

Lastly, FYI, TcUnit does not compile cleanly with the SFC checks:

```
Build complete -- 0 errors, 0 warnings : ready for download!
Additional code checks ...
FB_AdsAssertMessageFormatter.LogAssertFailure(9) : error: SA0167: Temporary FunctionBlock instance: 'AdjustAssertFailureMessageToMax253CharLength'
FB_TestSuite.GetTestByPosition(2) : error: SA0167: Temporary FunctionBlock instance: 'GetTestByPosition'
FB_TestSuite.AreAllTestsFinished(4) : error: SA0167: Temporary FunctionBlock instance: 'GetCurTaskIndex'
FB_TestSuite.AssertArray3dEquals_REAL(34) : error: SA0167: Temporary FunctionBlock instance: 'FormatString'
C:\Dev\CatFish\src\3rdparty\TcUnit\TcUnit\TcUnit\POUs\FB_TestResults.TcPOU(20) : error: SA0167: Temporary FunctionBlock instance: 'TestToBeStored'
FB_StreamBuffer.Clear(6) : error: SA0027: Variable name 'Count' in 'Clear' already used for an object in Library 'tc2_utilities, 3.3.47.0 (beckhoff automation gmbh)'
FB_TestSuite.AreAllTestsFinished(4) : error: SA0027: Variable name 'GetCurTaskIndex' in 'AreAllTestsFinished' already used for an object in Library 'tc2_system, 3.4.24.0 (beckhoff automation gmbh)'
FB_TestSuite.AssertEquals(13) : error: SA0027: Variable name 'Count' in 'AssertEquals' already used for an object in Library 'tc2_utilities, 3.3.47.0 (beckhoff automation gmbh)'
```

But

closed 2022-01-18 by marcusl